hzeller/rpi-rgb-led-matrix #880: Sending a fixed text

oculos, 2019-08-30:
Hi,

I've saw some other questions here on how to print a static text from the command line. Piping to text-example doesn't work, as the application quits - or seem to quit. 
Any other tip on how to do this? I would like to send a text from the command line, and I can control the process with pgrep if I need to kill it, but the only trouble I face is that I wish I could send the text without having to type it to STDIN.

oculos, 2019-08-30:
Would the --led-daemon option help on this? I didn't figure out how to use it.

hzeller, 2019-08-30:
The text example is just an example.
If you want a particular output it is the simplest to just take that code example and modify it to your needs, e.g. by using a text from the command-line instead of the input file-descriptor.