hzeller/rpi-rgb-led-matrix #884: Problem executing script with php page

eSenKaa, 2019-09-06:
Hi ! First of all, thanks for your awesome library.

I'm here because I have an issue, I made my own buildroot with sudo and I would like to execute a .sh from a php page. (The .sh works pretty well without sudo n my raspberry to execute things on my led screen) but when I try on my php page I got this in the logs :

Must run as root to be able to access /dev/mem
Prepend 'sudo' to the command

How can I manage to have it works without sudo.

Thanks in advance if you have an answer.

hzeller, 2019-09-06:
you can set `demo` SUID root.

eSenKaa, 2019-09-06:
thanks for your fast answer, could you be more specific with more details because i'm just a beginner so that don't talk much to me :/

hzeller, 2019-09-06:
use chmod to set this flag; use Google to understand more of the concept if you haven't encountered it before (sorry, on phone, can't easily write)

eSenKaa, 2019-09-06:
ok I found something and it actually works, thanks a lot :)

closed 2019-09-06 by eSenKaa