hzeller/rpi-rgb-led-matrix #886: Not an issue but a Doubt

vaseeece, 2019-09-08:
I'm using only one panel of 64x32 of pitch 4 mm and I want to connect a GSM module(sim800) with raspberry pi, unfortunately, the GSM hast o be connected with pins 8 and 10 on raspberry pi, Now this library is using the pin 8 for (R1) connection of panel 3. If I connect the GSM module there at pin8 in raspberry pi would it cause any error to hardware or library installed???  
If so, is there any other way that I could connect the GSM module and panel to the raspberry pi at the same time??

vaseeece, 2019-09-09:
@hzeller can you please help me with this and give me a solution for this, please........

hzeller, 2019-09-11:
If you only use one panel, you use `--parallel=1`, so the pins are free.

Pin8 (GPIO14) is only used by --parallel=3, so is not a problem here.

Pin 10 (GPIO15) is only used with a 64 pixel high panel with ABCDE address lines; you only have 32, so that is fine.

So you will not have problems with that, and the serial interface should still be free for your GSM module to use.

(You also can re-map GPIO pins used in the [hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c) file, but given that you only use one panel, this is not needed).

closed 2025-11-27 by marcmerlin