hzeller/rpi-rgb-led-matrix #889: Is there any way of displaying images from inside C++ code ?

GeorgeFlorian, 2019-09-11:
Hello !

Is there any way of displaying images using methods in a C++ program ?

To display a text line we use `rgb_matrix::DrawText()`. What do we use when we need to display an image ?

Can somebody point me in the right direction ?

G-Phong, 2021-08-08:
I have the same problem. I don't know how to show images/GIFs on the LED matrix per C++ Code. I've gone through utils/led-image-viewer.cc code but it has >500 lines of code and for me as a beginner it is not easy to understand everything.

I found the function

`void DisplayAnimation(const FileInfo *file, RGBMatrix *matrix, FrameCanvas *offscreen_canvas)`

but do not know how to use it properly to show a .png file or a GIF...

Anybody got ideas?

K0rkunc, 2021-10-17:
> Merhaba !
> 
> Bir C++ programında yöntemleri kullanarak görüntüleri görüntülemenin herhangi bir yolu var mı?
> 
> Bir metin satırını görüntülemek için `rgb_matrix::DrawText()`. Bir resmi göstermemiz gerektiğinde ne kullanırız?
> 
> Birisi bana doğru yönde işaret edebilir mi?

are u solved?

hzeller, 2021-10-17:
mmh, I guess I should add a simple image viewer example in `examples-api-use/`

K0rkunc, 2021-10-17:
> mmh, sanırım içine basit bir resim görüntüleyici örneği eklemeliyim `examples-api-use/`

im solved
  this example
w++;
		if(w==imgs.size()-1){w=0; off_canvas->Clear();}
			for (size_t y = 0; y < img0[0].rows(); ++y)
			{
				for (size_t x = 0; x < img0[0].columns(); ++x)
				{
					const Magick::Color &c = imgs[w].pixelColor(x, y);
					off_canvas->SetPixel(x, y,
										 ScaleQuantumToChar(c.redQuantum()),
										 ScaleQuantumToChar(c.greenQuantum()),
										 ScaleQuantumToChar(c.blueQuantum()));
										 
				}
			}
		
		rgb_matrix::DrawText(off_canvas,device_1.font1,50,50,mRed, NULL,"asd");
		

		// device_1.ImgPaint(off_canvas,"scene1rsz.gif",0);

		off_canvas = matrix->SwapOnVSync(off_canvas);

K0rkunc, 2021-10-17:
void ShowAnimatedImage(const ImageVector &images, RGBMatrix *matrix) {
  FrameCanvas *offscreen_canvas = matrix->CreateFrameCanvas();
  while (!interrupt_received) {
    for (const auto &image : images) {
      if (interrupt_received) break;
      CopyImageToCanvas(image, offscreen_canvas);
      **offscreen_canvas = matrix->SwapOnVSync(offscreen_canvas);**
      usleep(image.animationDelay() * 10000);  // 1/100s converted to usec
    }
  }
you print the canvas directly to the screen where I marked, actually it would be better if you remove it. It seems like a better idea to print the canvas to the screen at the end of the loop.

It is easier to add and edit text on a picture or animation.

K0rkunc, 2021-10-17:
> void ShowAnimatedImage(const ImageVector &images, RGBMatrix *matrix) { FrameCanvas *offscreen_canvas = matrix->CreateFrameCanvas(); while (!interrupt_received) { for (const auto &image : images) { [...]

> Exactlk @K0rkunc aslında sadece pikselleri tuvale kopyalıyor.
> 
> Şimdi example-api-use dizinine basit bir resim örneği ekledim: [https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/image-example. cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/image-example.cc)

void ShowAnimatedImage(const ImageVector &images, RGBMatrix *matrix) {
FrameCanvas *offscreen_canvas = matrix->CreateFrameCanvas();
while (!interrupt_received) {
for (const auto &image : images) {
if (interrupt_received) break;
CopyImageToCanvas(image, offscreen_canvas);
offscreen_canvas = matrix->SwapOnVSync(offscreen_canvas);
usleep(image.animationDelay() * 10000); // 1/100s converted to usec
}
}
you print the canvas directly to the screen where I marked, actually it would be better if you remove it. It seems like a better idea to print the canvas to the screen at the end of the loop.

It is easier to add and edit text on a picture or animation.

hzeller, 2021-10-17:
In the example, I want to highlight best practices. If you don't do the offscreen canvas, there can be tearing in the image, this prevents it.

But of course, if you just need a single image and no animation, no need for offscreen canvas.

K0rkunc, 2021-10-17:
thanks for share master 

open source receiver cards

These are very affordable, have you ever thought of writing applications to such receiver cards?

sorry if i wrote something wrong i use google translate :)