hzeller/rpi-rgb-led-matrix #891: Strange flickering

Maerchensarg, 2019-09-11:
Hello,

I'm just trying my luck to get a 64x64 LED panel to run. First I bought the "RGB Matrix Panel Drive Board For Raspberry Pi", but I guess it doesn't work with my panel (just two strings showing up, connected E to P8, and GND to P4).

The following LED panel was bought: https://s.click.aliexpress.com/e/rkbKqIv6
Unfortunately, I didn't pay much attention to the details beforehand and somehow assumed that it would work (naive -.-).

I'm using the RPI4 and have now done the wiring by hand (without the board).

I start the demo with it:
"sudo ./demo --led-rows=64 --led-cols=64 -D0"

The Python script:
"sudo ./simple-square.py --led-rows=64 --led-cols=64"

The result looks like this (there is also a picture of the panel):
https://imgur.com/a/hnFaQlH

I also played with several other commands, but without success.

Maybe someone has an idea? Unfortunately, I'm not very deep in the matter, but somehow obsessed with colourful light.

Thanks for your hard work and have a nice day.

coman3, 2019-10-02:
Ive got this exact same display module (same seller and everything), and have no problems with my rpi3. Maybe try the gpio_slowdown option of the examples?

Maerchensarg, 2019-10-05:
It can be that simple. Thanks for the hint, because now everything is working as planned. Thanks!

closed 2019-10-05 by Maerchensarg