hzeller/rpi-rgb-led-matrix #895: 64x64 LED Matrix w/ adafruit bonnet - black bars

dmathew93, 2019-09-14:
I just bought the 64x64 LED matrix and the adafruit bonnet recently and followed the instructions to assemble them appropriately. I am running into the issue where when I am running it in 64x64 mode large black bars will appear alongside 2 large blue bars. Included are the images of my setup and the display. I have done the E line from 8 to E, I have soldered the connection from 4 to 18 and tried a variety of options from the library owners github. Any help would be appreciated and would like to rule out if its the bonnet board itself.

https://imgur.com/a/kXnbrOB

GeekyTim, 2019-09-14:
Did you solder the bonnet?

dmathew93, 2019-09-14:
What do you mean by solder the bonnet? I soldered the E address lines and pin 4 to 18. Included an imgur link of my board.

GeekyTim, 2019-09-14:
That would be the one.

hzeller, 2019-09-20:
Your picture actually looks like the D line is not properly connected as all the rows with that bit set are black, so they are folded onto each other.
Try to follow the D line and see if there is any break on the circuit board or the cable.

dmathew93, 2019-09-21:
> Your picture actually looks like the D line is not properly connected as all the rows with that bit set are black, so they are folded onto each other.
> Try to follow the D line and see if there is any break on the circuit board or the cable.

Hey Hzeller, thanks for the reply.

I just too my multimeter and started measuring the voltage on each address line. You were right. The D address line is not showing any power whatsoever. All others are showing 1.6V. That explains why the black bars are there, but i dont see how it explains the two large blue columns. Any ideas on that one?

Denzel

dmathew93, 2019-09-21:
Trying to debug this issue with the D address line and the always on blue bars I’m now seeing this issue where the board is reacting to interference.

https://imgur.com/a/i4OSX9T

hzeller, 2019-09-21:
when lines are a solid color typically means that they are not driven properly. This could be due to cable issues or badly soldered connections on your bonnet.
You do set the right `--led-gpio-mapping` that you need for your set-up, right ?

dmathew93, 2019-09-21:
> when lines are a solid color typically means that they are not driven properly. This could be due to cable issues or badly soldered connections on your bonnet.
> You do set the right `--led-gpio-mapping` that you need for your set-up, right ?

Correct, I do set the mapping to the adafruit-hat one. I tried all others just to rule things out. I thought it could be the cable so I tried to measure voltages on just the board itself on the RGB pins and the B2 pin is driven to a 3V output. B1 is going to 0.

mariana-s-fernandes, 2021-01-16:
Hi @dmathew93, please check my comment [here](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1189#issuecomment-761565883) to see if that works for you.