hzeller/rpi-rgb-led-matrix #896: 32x32 1/8 scan rate outdoor panel not working

toma3757, 2019-09-15:
Hi,

I bought a LED panel and a [Raspberry PI hat ](https://www.acmesystems.it/HAT-A3) and I would like to make it work but currently I don't know how to display the demo layouts correctly.
The panel only displays something when I set the --led-parallel=2 parameter. (Maybe it is made from 2 pieces of 16*32 panels?)
I got these outputs:
--led-rows=16 --led-cols=32 --led-parallel=2 --led-multiplexing=1
[image]
--led-rows=16 --led-cols=32 --led-parallel=2 --led-multiplexing=0
[image]
--led-rows=16 --led-cols=32 --led-parallel=2 --led-multiplexing=2
[image]

I tried a lot of combinations of the available parameters without success.
I have read similar issues but their solutions not work work for me.
Can you suggest something to try?

hzeller, 2019-09-16:
Led parallel is only needed if you actually connected to two IDC outputs. But it is hard to tell: you have omitted the crucial information how you did the connection - [active adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) or [manual](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) ? Please provide a picture of your wiring, and picture of the IDC connector on the LED panel.

If it _only_ works with parallel=2, you somehow have connected your panel to the second chain and not he first. 

_Only_ use `--led-rows=16` and `--led-parallel=2` if your 32 row panel has two input IDCs connected.
If you have a single IDC, then always use `--led-rows=32 --led-cols=32`, not `--led-rows=16`, otherwise you confuse yourself by messing up the led-multiplexing.

If all the led-multiplexing (0..10) variants don't work for you, you have to figure systematically the mapping for your panel and provide one in [`lib/multiplex-mappers.cc`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc).

toma3757, 2019-09-17:
Thanks!
I tried a lot of combinations of parameters and connections(top/middle) but somehow I missed some of them.
I have to use --led-multiplexing=6 and the panel is connected to the top.

closed 2019-09-17 by toma3757