hzeller/rpi-rgb-led-matrix #898: Clock pulse duration required

OllmGrb, 2019-09-20:
Hi, I am using a 64x64 p2.5 display to try some demo programs ( one of the newer modules with an extra e pin) . I have followed the wiring diagrams but none of them seem to be working
Hence im gonna try write a simple program, hopefully understanding from scratch will help me. 
I understood the way the matrices work and what is required by all the pins, except for the clock and strobe pins. I understand that the clk pin requires a pwm signal, i just dont know the width required. As for the strobe pin, I have no clue what is required and how it affects the output. 

Any help would be greatly appreciated, thanks

hzeller, 2019-09-20:
If it is one of the newer 64x64 panels, maybe `--led-panel-type=FM6126A` helps.

Clock shifts new data into the shift register which is two RGB rows, in your case 32 pixesl apart. Strobe takes them over to be visible. `OE` switches that line on, so it is divided in bit-planes and the pulse sent is just doubling every bit-plane. To generate the pulse, it internally uses the hardware of the PI used for PWM, but the process really is what some people call Binary Code Modulation.

OllmGrb, 2019-09-20:
Ok, understood the sequence, thanks.
So the pwm is taken immediately from the pi, and no external real time clock ( like from the pi hat) is required, correct?

hzeller, 2019-09-20:
Yes, the RTC is only there to give the Pi a correct walltime at bootup as most Pi Zeros are not connected to the network and would not know what time it is.

OllmGrb, 2019-09-20:
Hey, just managed to run the first demo using the panel type like you said. Only thing, the demo runs for about 4 seconds, before the screen just switches on random leds as shown in the video. Loose connection issue or software? thanks
https://photos.app.goo.gl/D6cuKDHw1EJQsMKG7

hzeller, 2019-09-20:
Check the troubleshooting section. This looks like 1-wire connection was not disabled on your Pi.

OllmGrb, 2019-09-20:
Hi checked and you were right. Removed that line but screen still flickers. Was reading the trouble shooting section, any other obvious services that might be interfering with the demo?

hzeller, 2019-09-20:
If you run a minimal Raspian configuration without UI etc. there should usually not be anything else that messes with the GPIO lines.

Maybe the panel has trouble with the speed; try to increase the `--led-slowdown-gpio` value.

closed 2019-10-10 by OllmGrb