hzeller/rpi-rgb-led-matrix #900: crash on Pi 3 Model A+

TonivdW, 2019-09-29:
Sparkling new model 3A+, freshly installed Raspian Lite, fully updated.
Right after starting ft-server the screen is getting overwritten with green or purple blurbs and sometimes kernel panics are reported. 

Is FT supported on model 3A+?

[image]

hzeller, 2019-09-29:
If with FT, you mean https://github.com/hzeller/flaschen-taschen , then you probably have to update the submodule (it is in a version before the automatic A+ conversion). I'll fix that there.

hzeller, 2019-09-29:
I've updated the version now, can you do a submodule update or a fresh recursive update with FlaschenTaschen to see ?

TonivdW, 2019-09-30:
Yes, that fixed it! Many thanks.

closed 2019-09-30 by TonivdW