hzeller/rpi-rgb-led-matrix #901: P10 Outdoor 16x32 half matrix displayed green by Demo 0

lu93pi, 2019-09-30:
Hello,

I tried to control my LED matrix according to the instructions, which also works, but half of the matrix is only green (with demo 0)
See image
[image]

can somebody help me with this?

lu93pi, 2019-10-07:
I'm using Raspberry pi3 with an adafruit-hat and 16x32 Panel from LyssonLed..
 just executing the Demo 0 ..

lu93pi, 2019-10-07:
-D0 --led-gpio-mapping=adafruit-hat --led-rows=16 --led-cols=32 
thats all

hzeller, 2019-10-07:
if it is an outdoor panel, you might also need to play with the minute multiplexing options

lu93pi, 2019-10-07:
tried all multiplexing options .. 0-9 .. the top half of the display is still green

lu93pi, 2019-10-07:
tried another pi.. with the --led-rows=16 command the 4. 8. and 12. row dont work

davemaster, 2019-11-10:
What panel is it exactly?

closed 2025-11-27 by marcmerlin