hzeller/rpi-rgb-led-matrix #902: Extreme visual noise on all recently purchased LED panels

PaulSlocum, 2019-10-01:
I'm using 4mm 64x32 LED panels and all of the panels I've bought recently from Adafruit and multiple vendors on Amazon have a ton of visual noise compared to Amazon panels I bought earlier this year.  I was not able to find any timing or color parameter that fixed the issue.  I'm using the very short ribbon cables that came with the panels, and I've tried different cables. Using a 5V 10A power supply.

Here are the Amazon panels I bought earlier this year that always have a clear picture:
https://www.youtube.com/watch?v=4Ojo11erqdQ

Here are the Adafruit panels I got this week with a huge amount of noise, and every Amazon panel I've purchased recently has the same issue:
https://www.youtube.com/watch?v=zMdqvEBiD8A

Are these panels bad or is there some way to fix this? I'm using the "high quality" mode of the Matrix Hat with RPi audio disabled.

PaulSlocum, 2019-10-02:
I realized that overclocking the RPi GPU was contributing to some visual noise.  After removing the overclocking, now all the Amazon panels work with minimal noise, but the lot I bought recently from Adafruit still look just as noisy as the video I posted.

jaygarcia, 2019-10-24:
Some thoughts and questions:
- What pis are you using?
- Have you disabled audio?
- Perhaps trying this without HDMI video? (just a thought) :)
- Have you tried `--led-slowdown-gio`?   have had similar issues.
https://github.com/hzeller/rpi-rgb-led-matrix/issues/861

PaulSlocum, 2019-10-26:
I forgot to update, it ended up needing "--led-slowdown-gpio=2"

closed 2019-10-26 by PaulSlocum