hzeller/rpi-rgb-led-matrix #904: Can I set individual chain length ? Will one RPI 4 be enough for 50 panels ?

GeorgeFlorian, 2019-10-04:
Hello !

I have a project that involves a big display, composed of 5x10 panels.

I would like to know if I can do this with only one RPI 4 and, since 50/3=16.6, what should the `chain_length` value should be ?

I am planning to chain them like this:
```
   [<][<][<][<][<]  }--- Pi connector #1
   [>][>][>][>][>]
   [<][<][<][<][<]
   [>][>][>][>][>]
   [<][<][<][<][<]  }--- Pi connector #2
   [>][>][>][>][>]
   [<][<][<][<][<]
   [>][>][>][>][>]
   [<][<][<][<][<]  }--- Pi connector #3
   [>][>][>][>][>]
```

First and second chain would have 20 panels each and the last one will have 10. I will use `pixel_mapper_config = "U-mapper";`.

What do you think ?

Thank you !

shanerooni, 2020-03-06:
did you ever figure this out?  i'm trying to create a chain of 12 panels in a 3 rows x 4 columns configuration with 32x64 panels.  i read something about an "s-mapper", but i'm also reading that "parallel" only works with the chip board that has 3 outputs.  i'm using an adafruit hat, so i need to run everything off of 1 continuous chain of 12.  thanks!

hzeller, 2020-03-06:
you want to run them parallel using one of the parallel adapters, the single chain adafruit hat will result in too low of a refresh rate for 12 panels. You can test the expected refresh rate by running the program with the necessary options and outputting the refresh rate.

shanerooni, 2020-03-06:
i actually am running all 12 at once right now, and don't see too much of a refresh rate issue.  i'm just using it for a MM:SS timer for a kids running series, and maybe to scroll a few announcements.  i snaked them as shown above, basically:

<<<< - pi4
>>>>
<<<<

i'm just not sure how to get it to read the panels in the correct way with the mapping.  where can i find an adapter?  does it take the place of the adafruit hat on the 40 pins?  thanks!

shanerooni, 2020-03-06:
This is a quick video with the image scroller. I have some flicker, and I'm
planning to write the two leads together I saw mentioned in another post.

hzeller, 2020-03-06:
Check out the page that describes the adapters https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter
There is also a 3rd party provider that sells fully built 3 chain connectors for about $3.

shanerooni, 2020-03-07:
is there a way to create an S-mapper with the adafruit hat?  i'm willing to give it a shot and see how the display looks before venturing to a separate board.

shanerooni, 2020-03-09:
sorry to be persistent, but is there a way to do this without the adapter if i'm okay with the low refresh rate?  i just need to invert those middle 4 panels.  thanks!