hzeller/rpi-rgb-led-matrix #905: Exception: No module named core

TenchiMuyo1984, 2019-10-04:
Hello, I've a problem executing the samples fpr python.

I get everytime this exception:
`PS D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\samples> python runtext.py
Traceback (most recent call last):
  File "runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\samples\samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\rgbmatrix\__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
PS D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\samples>`

I was thinking, that the core.pyx has a wrong ending and I changed from .pyx to .py
Try again and I got this exception:

`PS D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\samples> python runtext.py
Traceback (most recent call last):
  File "runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\samples\samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\rgbmatrix\__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
  File "D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\rgbmatrix\core.py", line 4
    from libc.stdint cimport uint8_t, uint32_t, uintptr_t
                           ^
SyntaxError: invalid syntax
PS D:\Downloads\Chrome\rpi-rgb-led-matrix-master\bindings\python\samples>`

What I do wrong?

My target is to run a 32x64 with FM6124 on it with a RasPy3b+ and python (2.7).

hzeller, 2019-10-04:
the paths mentioned in that error message indicate that you're running this on some windows-based system? You want to compile that on your Pi.

TenchiMuyo1984, 2019-10-04:
For this example, yes. It was a traceback from my Win10.
But I got the same on my RasPy.

I'm able to execute the demo by using: 
`sudo examples-api-use/demo -D0`

`pi@raspberrypi:~/rpi-rgb-led-matrix-master $ sudo examples-api-use/demo -D0
Size: 32x32. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs`

Sadly I've two squares rotating on my display....
But how can I execute a python code only?
I wanted to try this:
`pi@raspberrypi:~/rpi-rgb-led-matrix-master/bindings/python/samples $ python runtext.py'
but got this:

'Traceback (most recent call last):
  File "runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/rpi-rgb-led-matrix-master/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/rpi-rgb-led-matrix-master/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
pi@raspberrypi:~/rpi-rgb-led-matrix-master/bindings/python/samples $`

Or here are no examples in python?

hzeller, 2019-10-04:
If you have two squares: do you happen to have a 64x32 panel ? Then you need to give it `--led-cols=64` parameter.

As for Python: Please follow the instructions to install it as described in the readme https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python

TenchiMuyo1984, 2019-10-05:
Thank you for your time!

> If you have two squares: do you happen to have a 64x32 panel ? 
Yes, look in the last line of my first post ;)

>Then you need to give it `--led-cols=64` parameter.
Thank you, that was it!
 
> As for Python: Please follow the instructions to install it as described in the readme https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python

I've done this, but:
`sudo apt-get update && sudo apt-get install python2.7-dev python-pillow -y`
Was working fine.
`make build-python`
Didn't worked at the first try. I needed to change the directory. 
`sudo make install-python`
Same problem as before. I don't know why I must change the directory ...

But now, It works to execute the .py examples *thumbs up*

And maybe I'm stupid, but why I must use "sudo ./" at beginning? I use normaly "python demo.py" and that was it.

As Information. When I execute this code, I got a messy picture:
`sudo ./image-scroller.py --led-chain=2 --led-slowdown-gpio 0`

A new question:
Is it possible to change (decrease) the frequency to see a really slow buildup of the objects (pictures, symbols, etc.)?

arahasya, 2020-01-25:
hello. How did you run FM6124. I have the  same panel but not getting correct output

PoojaPremnath2003, 2023-09-02:
@TenchiMuyo1984 - If I may ask, which directly did you change to- before executing the make build-python statement? I'm facing the same issue now, and am trying to fix it.