hzeller/rpi-rgb-led-matrix #906: RGBLedMatrixOptions.GpioSlowdown property has no effect (C# binding)

daniel-richter, 2019-10-04:
Setting the `RGBLedMatrixOptions.GpioSlowdown` property has no effect. It always has the default value of 1.

In my opinion, this should never have worked before. While options like cows, cols, chain length, etc are part of `RGBMatrix::Options`, gpio-slowdown is part of `RuntimeOptions` (and is never passed to `led_matrix_create_from_options`).

My current workaround is to hardcode set `DRGB_SLOWDOWN_GPIO` in `lib/MAKEFILE`

hzeller, 2019-10-04:
can you provide a pull request that makes it work?

daniel-richter, 2019-10-04:
I'll try, but I'm not sure what the best way to do this is, though. Adding a `led_matrix_create_from_options_2` that takes RuntimeOptions as a parameter? RuntimeOptions seem to be read directly from the command line at the moment...

hzeller, 2019-10-05:
As a quick workaround for now, could you just pass the command-line option to your program ? Or pass it to the argc/argv manually when invoking `led_matrix_create_from_options`

wibbleboy, 2019-11-05:
I'm glad that I found this thread; I hit the same issue yesterday where the Python examples were working on my Pi4, but I was getting ghosting of the pixels on my third panel whenn using the c# library, regardless of the slowdown property.

Building Daniels's workaround branch allows me to fix this by setting the slowdown to 5 via RGBLedMatrixOptions. Thanks.

PR #921 (2019-12-24): fixed GpioSlowdown parameter passing
* _GPIO slowdown_ is part of `RuntimeOptions` (and will never be passed to `led_matrix_create_from_options`)
* pass it as `--led-slowdown-gpio` to `argc`/`argv` manually when invoking `led_matrix_create_from_options`

this commit fixes #906
commit d5624e70 (2019-12-24): C# binding: fixed GpioSlowdown parameter passing (#921)

  * gpio-slowdown is part of RuntimeOptions (and will never be passed to led_matrix_create_from_options)
  *  pass it to argc/argv manually when invoking led_matrix_create_from_options
this commit fixes #906
closed 2019-12-24 by hzeller