hzeller/rpi-rgb-led-matrix #907: video viewer error

vaseeece, 2019-10-05:
When I try to make the video-viewer in utils after installing dependencies as mentioned in utils readme it shows error and not getting build.

please help out...

error details are given below...

root@DietPi:/home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils# make video-viewer
make -C ../lib
make[1]: Entering directory '/home/dietpi/RGB_matrix/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/dietpi/RGB_matrix/rpi-rgb-led-matrix/lib'
g++ -Wall -O3 -g -Wextra -Wno-unused-parameter -D_FILE_OFFSET_BITS=64 video-viewer.o -o video-viewer -L../lib -lrgbmatrix -lrt -lm -lpthread `pkg-config --cflags --libs  libavcodec libavformat libswscale libavutil`
/bin/sh: 1: pkg-config: not found
/usr/bin/ld: video-viewer.o: in function `main':
/home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:157: undefined reference to `av_register_all'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:158: undefined reference to `avformat_network_init'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:161: undefined reference to `avformat_open_input'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:165: undefined reference to `avformat_find_stream_info'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:211: undefined reference to `avcodec_find_decoder'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:217: undefined reference to `avcodec_alloc_context3'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:218: undefined reference to `avcodec_copy_context'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:224: undefined reference to `avcodec_open2'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:228: undefined reference to `av_frame_alloc'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:231: undefined reference to `av_frame_alloc'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:236: undefined reference to `avpicture_get_size'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:238: undefined reference to `av_malloc'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:243: undefined reference to `avpicture_fill'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:247: undefined reference to `sws_getContext'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:277: undefined reference to `av_read_frame'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:288: undefined reference to `avcodec_decode_video2'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:312: undefined reference to `av_free_packet'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:325: undefined reference to `av_free'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:326: undefined reference to `av_frame_free'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:329: undefined reference to `av_frame_free'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:332: undefined reference to `avcodec_close'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:333: undefined reference to `avcodec_close'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:336: undefined reference to `avformat_close_input'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:293: undefined reference to `sws_scale'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:273: undefined reference to `av_seek_frame'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:274: undefined reference to `avcodec_flush_buffers'
/usr/bin/ld: /home/dietpi/RGB_matrix/rpi-rgb-led-matrix/utils/video-viewer.cc:170: undefined reference to `av_dump_format'
collect2: error: ld returned 1 exit status
make: *** [Makefile:31: video-viewer] Error 1

hzeller, 2019-10-20:
The error message sais in the beginning that it didn't find `pkg-config`. So next step would be to install pkg-config.
```
sudo apt-get install pkg-config
```

hzeller, 2019-10-20:
(I presume you did install the packages needed as described in the [README](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer) ?)

rijotech, 2020-02-07:
I'm facing similar issue when compiling 
sudo make video-viewer
g++ -I../include -Wall -O3 -g -Wextra -Wno-unused-parameter -D_FILE_OFFSET_BITS=64 -Wno-deprecated-declarations -c -o video-viewer.o video-viewer.cc
make -C ../lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
g++ -Wall -O3 -g -Wextra -Wno-unused-parameter -D_FILE_OFFSET_BITS=64 video-viewer.o -o video-viewer -L../lib -lrgbmatrix -lrt -lm -lpthread `pkg-config --cflags --libs  libav        codec libavformat libswscale libavutil`
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libavcodec.so: undefined reference to `bcm_host_is_fkms_active'
collect2: error: ld returned 1 exit status
make: *** [Makefile:31: video-viewer] Error 1

rijotech, 2020-02-07:
Found the solution it was relate to the kennel, just perform an upgrade and try again. 
sudo apt update && sudo apt upgrade