hzeller/rpi-rgb-led-matrix #908: Individual control

OllmGrb, 2019-10-10:
HI, im working with python and need control of each individual led. I took a look at the graphics.py sample, and stumbled upon the drawline function, which i could use  to obtain what i require, 
I followed its path to the graphics.cpp , and found where the function is being declared. however, in the attempt to create a function from scratch , there are some lines of code i cant quite grasp, hopefully you can shine some light on the situation.

1. cyton_fallthrough
2. pytuple_get_item

note - a simpler way would obviously be to use drawline and give both x and y co ordinates the same, was just curious to see If there is a simpler way of controlling leds than writing a funtion from scratch. thanks

hzeller, 2019-10-10:
Why does `matrix.SetPixel()` not work for you ?

If you want to implement something on the C++ side and use in Python, you need to read yourself into the Cython documentation. The functions you mention look like auto-generated from it.

OllmGrb, 2019-10-12:
Wasn't aware of that command, thanks for the help. Read through the documentation and got a brief idea of what is what. thanks again

closed 2019-10-12 by OllmGrb