hzeller/rpi-rgb-led-matrix #910: P4 outdoor panel 32x64 - not working lower half of display

misudes, 2019-10-17:
I have problem to display lower part of LED P4 outdoor panels (64X32 Pixels, 1R1PG1GB SMD 1921, 1/8 scan, refresh frequency >=400Hz, www.led-card.com) with my AdafruitHAT and RPI 3B+.
I have for the test purpose in a chain two pannels only.

This working well in upper half of display (scrolling ppm):
sudo ./demo -D1 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-chain=4 --led-multiplexing=6 runtext16.ppm
[image]

For runtext.ppm is not working.
sudo ./demo -D0 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-chain=2 --led-multiplexing=6 runtext.ppm
[image]

sudo ./demo -D0 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=32 --led-chain=4 --led-multiplexing=6
[image]

I try E-link (pin connections to 8 or 16) without any change.
I try resolderin pins on AdafruitHAT, all seems to be fine.

misudes, 2019-10-18:
I have datasheet info from panel supplier for my P4 outdoor LED panel.
Constant current 1/8 scan
Driver chip: ICN2037
Decoding chip: 7258
No E-Line

misudes, 2019-10-19:
sudo ./demo -D0 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=32 --led-chain=2 --led-multiplexing=6
[image]

sudo ./demo -D3 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=32 --led-chain=2 --led-multiplexing=6
[image]

Hi everyone. I try a lot of combinations with only one result you can see above. Lower half part is not addressing. Any ideas how to working with these type of panel?
I know it is construct from 2-sub-panels, there for -led-chain=2.
It is problem in driver chip ICN2037? New one in this library and it is difficult to drive the panel?
I bought theme in 8/2019.
I am working on a project and I have right now a big STOP with these panels :(

--led-inverse
It seems to be working - the whole LED panel area :)
[image]

Thank's for any help.
I am reading tons of other forums and issues here with no result what to do.

misudes, 2019-10-19:
Solved here: #311

For me works: --led-multiplexing=6

WernerFettke, 2025-08-30:
Hi, I have the exact same panels, but I'm not getting any output at all, not even jibberish . . . did your panels work straight out the box, or did you have to do anything special to get yours to work ? Any assistance would be appreciated.

Many thanks

Werner

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin