hzeller/rpi-rgb-led-matrix #918: Getting high refresh rates on a 192x128 matrix

carlosalaniz, 2019-10-29:
I'm driving a 192x128 matrix, however, the RPI can only go up to 60 ~ 100hz when playing an animation.

Chinese controllers can achieve high refresh rates 960hz+, however, these cards do not provide a good interface to be controlled programmatically, I need to be able to control individidual pixels on the matrix, stream video, among other things, that's why I'm looking into this library, anything that you would recommend to accomplish this at high refresh rates?

Thank you!

hzeller, 2019-10-29:
You need to give a little more detail about your setup.

What are the panels used, how many parallel and chains. How you do the animation? pwm depth?
What are the parameters you give to the library? What is the multiplexing? are you using pwm dither?

Cellgalvano, 2019-11-24:
@carlosalaniz These chinese controllers use FPGAs to achieve high refresh rates, they are doing the whole refresh in hardware. I do not know if one Pi is fast enough for such a big panel, if it isn't you could look into using a FPGA Hat for the Pi.

marcmerlin, 2019-12-30:
@carlosalaniz @Cellgalvano   it just so happens that I also setup a 192x128 matrix, and it was ok enough in some setups.
~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-row-addr-type=0 --led-chain=3 --led-show-refresh --led-slowdown-gpio=1 --led-parallel=1 --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=9 --led-pwm-bits=4 -D0
gives 390Hz, but --led-pwm-bits=4 isn't great

~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-row-addr-type=0 --led-chain=1 --led-show-refresh --led-slowdown-gpio=1 --led-parallel=3 --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=9 --led-pwm-bits=7 --led-panel-type=FM6126A -D10
aso gives 350Hz or so but with 7 bits of color (note that I had to patch the library to allow  --led-pwm-lsb-nanoseconds=9 which gives a small boost, not sure if @hzeller  would take a patch to remove the restriction that currently restricts the speed to 50ns).
Note that this 2nd invocation uses the parallel board with 3 outputs.

That was with ABCDE panels. @hzeller was telling me something about AB (or AC) panels being quicker than ABCDE panels. I was not sure how that is the case, since with 2 address lines, you have to feed 8 times more pixels per line, vs feeding 8 times fewer over 8 times more addresses.
I've noticed that 64x64 with 2x ABCD 64x32 panels where you shift 64 pixels per line is slightly faster than a single ABCDE panel with one more address line but you don't feed as many pixels, but not by much.
@hzeller  can you explain how AB/AC panels are supposed to be much quicker than ABCDE panels? I do need this for an upcoming project that will use 128x192 and would of course like to get the best panels with the best refresh rate if possible.

hzeller, 2019-12-31:
Yes, panels with ABCDE are super-slow as they have to do the relatively long PWM cycle 32 times. It is faster clocking in double the LEDs in a row and have half the multiplexing.

Look for 'outdoor' panels with 1:4 or 1:8 multiplexing. They are bright and fast.

What you did is good: always use as many parallel chains as possible, as it doesn't cost more to operate more chains, but the shorter they are, the faster they can go.

Don't go too low with the nanosconds speed, the panels can't switch that fast and the cabeling would also not properly transfer that pulse. Also, it doesn't buy you as much as the on-time is used to clock in the data in parallel; ultimately, you'd just be blocked by the clocking. Lower than 50ns has created issues in the past, so this is why it is limited; you're of course free to set this if it works for you.

Your invocations don't looks like you run them as root; might just be left out in the message, but if not, always run as root; this is required for hardware timing, proper realtime threads etc.

You can try this by looking what speeds you'd get if you give `--led-multiplexing=1` (the output on your panel will not be useful, but you'd see the speed).

marcmerlin, 2020-01-01:
Hi @hzeller  thanks for the tips.
Yes, I noticed that --led-pwm-lsb-nanoseconds=9 makes brightness go down, I just took it as long as it would reliably work on my panels. It does not increase the speed by a lot, but still does give some extra Hz.
Running as root: I definitely run as root, but thanks for making sure that I indeed do.
--led-multiplexing=1  does not make things slower or faster.
--led-row-addr-type=3 actually makes things slower, which goes against what you were saying (245Hz with vs 310Hz or so without)
```
oot@rPi3:~/rpi-rgb-led-matrix# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-row-addr-type=0 --led-chain=1 --led-show-refresh --led-slowdown-gpio=1 --led-parallel=3 --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=9 --led-pwm-bits=7 --led-panel-type=FM6126A -D10 --led-row-addr-type=3
```

I tried chain=3 parallel=1 for speed testing
```
root@rPi3:~/rpi-rgb-led-matrix# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-row-addr-type=0 --led-chain=3 --led-show-refresh --led-slowdown-gpio=1  --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=9 --led-pwm-bits=7 --led-panel-type=FM6126A -D10 --led-row-addr-type=3
```
gives 100Hz with type=3 and 110Hz with the default type.
Does this make any sense to you? For now, this tells me that AC panels will actually be slower than ABCDE panels :-/

marcmerlin, 2020-01-08:
@hzeller any idea why AC seems slower than ABCDE in the tests I just posted above?
Can you reproduce using the same command lines as the ones I posted?

hzeller, 2020-01-08:
I suspect because you are cheating with the very short pwm nanoseconds, so the clock-times dominate the timings.

marcmerlin, 2020-01-19:
@hzeller ok, let's take that --led-pwm-lsb-nanoseconds=9 and try -led-row-addr-type=3 in parallel or not. It's about 100Hz either way (huge slowdown)
```
root@rPi3:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-chain=1 --led-show-refresh --led-slowdown-gpio=1 --led-parallel=3 --led-pwm-dither-bits=1  --led-pwm-bits=7 --led-panel-type=FM6126A -D10 --led-row-addr-type=3
81Hz
```
```
root@rPi3:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-chain=3 --led-show-refresh --led-slowdown-gpio=1  --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A -D10 --led-row-addr-type=3
122hz max
```

Now, let's do a regular ABCDE panel, and it's still faster, 133hz
```
root@rPi3:~#  ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-row-addr-type=0 --led-chain=3 --led-show-refresh --led-slowdown-gpio=1  --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A -D10
133hz
```
Are you able to reproduce this?
If so, would you agree that ABCDE panels seem faster than AC panels at least the way that your driver is using them?

hzeller, 2020-01-19:
it is the panels with different multiplexing that make things faster, ABCD instead of ABCDE (64 row panel), or AB instead instead of ABC (16 row panels). That is what outdoor panels do.

It is _not_ the row address type you are testing here. That is just another way of doing the same 1:32 multiplexing so will not have much of a difference.

marcmerlin, 2020-01-19:
@hzeller 
Ah, thanks for clearing that up. So basically ABC panels are no better than ABCDE panels, gotcha.
Now I need to see if I can find AB panels that do 128x64, except it seems worse in the test I just did.

Just to be clear, 
--led-row-addr-type=3 is ABC
--led-row-addr-type=1 is AB
correct?

```
~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-chain=1 --led-show-refresh --led-slowdown-gpio=1 --led-parallel=3 --led-pwm-dither-bits=1  --led-pwm-bits=7 --led-panel-type=FM6126A -D10 --led-row-addr-type=1
```
is giving me only 94Hz for AB, which is even worse.

Am I doing something wrong?   Given that this can be tested without the actual panels, are you able to verify this from your side?

hzeller, 2020-01-19:
As I said, `--led-multiplexing` is the relevant factor, not `--led-row-addr-type`. The former decides what kind of multiplexing is done (and the layout within), which is in the ABCDE case 1:32. This needs to be less.

You need to look for so-called 'outdoor' panels, as they usually have a smaller multiplexing (and also thus are typically faster). However, I have seen 32x32 panels with outdoor, but not 64x64 yet. So it might be worthwhile looking if a longer chain with 32x32 outdoor panels can give any advantage.

The row-addr-type is only determining the way the multiplex address is given to the panel; it will not change much if you have a 1:32 panel.

Also: To compare times, I'd always start out with the defaults for pwm-bits and lsb-nanoseconds and then compare how that changes things when you tweak them (because you ideally would like to have a fast thing to start with and then tweak it later). Right now, you are setting them to extremes that are not really useful in real world but skew your comparisons as they drive things into 100% CPU.

marcmerlin, 2020-01-24:
@Cellgalvano you mention FPGA solutions which I know about, but I've never actually found the maximum number of pixels, or aggregate solution you can run from an FPGA.
From what I understand, it's not just about pushing pixels faster, ultimately if you refresh too quickly, lines will become dimmer as they are cycled too quickly.
Do you happen to have numbers for ABCD(E) panels vs AB panels?
@hzeller or maybe you also happen to know if rPi is ultimately already as fast as it can given the technology, and 256x256 is probably a reasonable limit if you want a 100Hz refresh rate, or whether FGPAs can go much higher somehow?

hzeller, 2020-01-24:
rpi is actually too fast for the panels which is why we have to slowdown gpio. I have something brewing which allows to multiplex more chains, but it is a project whenever-there-is-time, so don't hold your breath.

With FPGA's, you will always get better results as you have more io to connect more parallel chains. Clocking speed will not be much different (matrix limit), but easier to tune to optimum. Also they will be brighter as there is some dead-time in the timing circuit in the PI.

marcmerlin, 2020-01-24:
Thanks @hzeller for the details.  
I did some quick benchmarking with configuring AB panels I don't have, but as you said, I should be able to benchmark this before getting some, by running the output as if I were plugged in.

I think it could be useful to post some benchmark numbers to give people an expectation of maximum display sizes they can expect now that getting 256x256 is actually very achievable hardware-wise.

Did you do any blind benchmarking yourself of what's the maximum resolution that can be reasonably driven from an rPi3 while still keeping a refresh rate of 100Hz? (or whatever number you deem acceptable).

If not, I did some below. Here's my test for 128x192 on 3 channels without the performance increasing options:
```
demo --led-rows=64 --led-cols=128  --led-chain=1 --led-show-refresh --led-parallel=3 -D10
```
gives 100Hz

Adding --led-row-addr-type=1 (for AB) keeps the speed the same, as you hinted.

Using  --led-multiplexing=1-4 increases refresh to 160-180Hz, so that's a sizeable increase, but that increase goes away if I add "tuning options" (which indeed do decrease the quality, but are necessary to get a refresh rate that's high enough).

Once I add "--led-pwm-dither-bits=1 --led-pwm-bits=7" then the speed is 200Hz vs 320Hz for AB vs ABCDE

Adding --led-pwm-lsb-nanoseconds=9 then gives me around 430Hz in both cases of multiplexing.

AB panels are still a bit faster, but not anymore if you push --led-pwm-lsb-nanoseconds to a value is arguably somewhat undesirable as the panels get a bit too dim.  
Back to a more reasonable number (25) the difference is really just 10Hz between the 2:
```
~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=64 --led-cols=128  --led-chain=1 --led-show-refresh --led-parallel=3 -D10  --led-row-addr-type=1 --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=25 [--led-multiplexing=4]
```

Hope this helps someone. Looking forward to the first person to build a 256x256 and say how bad it looks, although ultimately, as you say, a lot of parallel strings with an FPGA (or multiple), is the only way to go if you want to go higher res and keep quality (seems that 128x128 at high quality is already too big for a single string).

marcmerlin, 2020-01-24:
There you go, make a README patch: https://github.com/hzeller/rpi-rgb-led-matrix/pull/971

daveythacher, 2020-08-07:
> I'm driving a 192x128 matrix, however, the RPI can only go up to 60 ~ 100hz when playing an animation.
>
> Chinese controllers can achieve high refresh rates 960hz+, however, these cards do not provide [...]

See my posts in #971 and #969

Cellgalvano, 2020-08-07:
My first guess would be to use the tipps and tricks from the previous posts. If you reached the limits of the pi you could look into the beaglebone project octoscroller or try to get a different firmware for your chinese controller board. Most of these boards feature FPGAs. Some boards like the ColorLight 5A-75B are well documented.

daveythacher, 2020-08-07:
> ColorLight 5A-75B

There could be a couple challenges with using the ColorLight 5A-75B. The primary interface is expected to be a custom ethernet protocol. I am not sure if it is standard L2 messages or not. However the 74HC245T drivers will likely block input on external pins. This was worked around with this project: https://hackaday.io/project/174032-orchid

marcmerlin, 2020-08-08:
If that inspires anyone, just finished the frame for my 384x256 array
[image]

While you can go a bit farther (384x320, or if you really push 384x384), those are probably the actual limits for the current library code. I'm sure the panels will go faster with custom  FPGAs, but as others have said, I haven't found any cheap FPGA that takes easy input you can generate from an arduino/Pi-like chip.

Build info: http://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html

daveythacher, 2020-08-09:
> If that inspires anyone, just finished the frame for my 384x256 array
> [image]
>
> While you can go a bit farther (384x320, or if you really push 384x384), those are probably the actual limits for the [...]

What commands did you have to use for this? What was the CPU usage? What other software is running in the background? Did you use all three chains?

The controller boards that are normally paired with these are fairly low cost. Especially for this generation as the bandwidth of the SPI drivers limits the memory required on the controller. FPGAs can handle this work load at full performance with small amounts of LUT resources, however memory directly scales with pixel density.

The two versions of the second generation panels improves the pixel density against IO bandwidth. So for those the controller can become a limiting factor. However they make many different sizes. Note most of this assumes high degrees of color depth. Things change a little in terms of memory footprint for that. It is possible the footprint rises and the pixel density increases.

Most of the panels and controllers are very cost optimized. They also use custom implementations and protocols. Some parts are designed to work against 5V. The controllers, power supplies and panels are all standardized. This reduction of components allows scaled production with corresponding cost reductions.

A third generation could exist which changes the panels or controller. It is possible to integrate a small slave controller into the panels which improves pixel density. This controller in theory could work externally to the internet directly. However this has a few hurtles. Currently most protocols have one to one mapping of bandwidth. In the master controller this does not need to happen. Meaning you could begin to use video streaming compression. However the slave to LED drivers there will still be a one to one.

The current technology uses the receiver as the slave controller, while the transmitter as the master. Perhaps they integrate the master in and change the master interface for source data. At this point you will start to see the possibility for security bugs which may already exist. However they could use Linux here, and this is where some subsystem considerations start to factor in.

In summary FPGAs have complexity against memory. Not a huge issue but can make them costly to make for low volume. Most of that likely in assembly. Now if you are willing to reduce the color depth, then things change. This reduces the bandwidth. 

See issue #1038 where I outline a small cheap FPGA. This in theory can do 128x32 @ 24 bit with 47 FPS. This one has a few interesting considerations. It uses SPI for interface to drive in row data, which depending on color depth format, places limit on column length. This is in part due to the SPI clock speed and the internal memory which is only about 1024x48/54 or 6144x8/9, with the possibility to increase to 7168x8/9. However by design a single row will only have half that as one row written while the other is read.

This FPGA would be very cheap to make. It could be built off TinyFGPA A2, which appears to be out of stock. The issue here is these use speed grade 4 which is a little slow for full HUB75 performance. My current work in progress would need speed grade 5 or 6. This adds at most a dollar or two.

As I outline in my other post this approach is not without issue. However I would view this as no worse than the current issues here with this project. You could go fix which will fast and more stable, however it will have the the issues I mentioned before. You could make the logic configurable to the point it could work on a number of FPGAs. Then you would just need the correct one. I actually looked into this briefly but never finished it. (I think I have that one archived, and it is a buggy mess. Like all my archived projects.)

The question here is internal vs external memory as you can save money with that. External memory tends to be single port which will increase board space and component cost. However this may still end up cheaper. There are several development board which can allow large FPGAs for low cost. Some development/breakout boards off nice price points for evaluation purposes. Others not so much.

daveythacher, 2020-08-09:
> Thanks @hzeller for the details.
> I did some quick benchmarking with configuring AB panels I don't have, but as you said, I should be able to benchmark this before getting some, by running the output as if I were plugged in.
> 
> I think it could be useful to post some benchmark numbers to give people an expectation of maximum display sizes they can expect now that getting 256x256 is actually very achievable hardware-wise.
> 
> Did you do any blind benchmarking yourself of what's the maximum resolution that can be reasonably driven from an rPi3 while still keeping a refresh rate of 100Hz? (or whatever number you deem acceptable).
> 
> If not, I did some below. Here's my test for 128x192 on 3 channels without the performance increasing options:
> 
> ```
> demo --led-rows=64 --led-cols=128  --led-chain=1 --led-show-refresh --led-parallel=3 -D10
> ```
> 
> gives 100Hz
> 
> Adding --led-row-addr-type=1 (for AB) keeps the speed the same, as you hinted.
> 
> Using --led-multiplexing=1-4 increases refresh to 160-180Hz, so that's a sizeable increase, but that increase goes away if I add "tuning options" (which indeed do decrease the quality, but are necessary to get a refresh rate that's high enough).
> 
> Once I add "--led-pwm-dither-bits=1 --led-pwm-bits=7" then the speed is 200Hz vs 320Hz for AB vs ABCDE
> 
> Adding --led-pwm-lsb-nanoseconds=9 then gives me around 430Hz in both cases of multiplexing.
> 
> AB panels are still a bit faster, but not anymore if you push --led-pwm-lsb-nanoseconds to a value is arguably somewhat undesirable as the panels get a bit too dim.
> Back to a more reasonable number (25) the difference is really just 10Hz between the 2:
> 
> ```
> demo --led-rows=64 --led-cols=128  --led-chain=1 --led-show-refresh --led-parallel=3 -D10  --led-row-addr-type=1 --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=25 [--led-multiplexing=4]
> ```
> 
> Hope this helps someone. Looking forward to the first person to build a 256x256 and say how bad it looks, although ultimately, as you say, a lot of parallel strings with an FPGA (or multiple), is the only way to go if you want to go higher res and keep quality (seems that 128x128 at high quality is already too big for a single string).

I question 9ns on the wire, maybe noise. Many LED drivers do not respond on CLK or OE at 111MHz. I am guessing this is the software generation frame rate.

I backed away from the fixed FPGA cause of the OE signal being not constant. Granted I could set a counter, but I decided to try to move over to a software solution to this. That FPGA library thing would have been cool. I could probably still work around it but it may need a few different FPGA images for stuff like weird multiplexing. That or trying to explain how to configure it, I could make the FPGA configurable too, but that may end up a complete mess. Maybe bootloader or partial reconfiguration...however that could force me into expensive parts/crazy workflow. I did consider configuration script which manages build options/flavor then compile it.

If you have ever seen how the FPGA controller software handles it, they force you declare your board profile. Then allow you to configure speed settings and tell you what you get. At least this way it is a complete surprise right.

Cellgalvano, 2020-08-09:
One small board that is definitely capable enough to power one chain is the MAX1000 or CYC1000. At such a low cost per board probably one per chain would be feasible. These feature 8MB of SDRAM. I tried to implement a simple driver for these, the only downside is the relatively low numbers of io pins.

daveythacher, 2020-08-09:
> CYC1000

Did you use the external RAM? That is a single 16 bit single port SDRAM, which would mean you would need to triple the clock feeding it for 24 bit RGB. This could be mean 150 MHz clock for full performance. Did you arbitrate with FIFOs? The ColorLight 5A-75B appears to use two 16 bit single port SDRAMs, but they are likely arbitrating quite a bit. 

I personally have favored using the internal memory of the FPGAs. These allow my to keep my clocks low. I can also avoid having to arbitrate, but I only have so much to work with.

These are listed at 402 MHz in clock tree, but that can be lowered by routing delays. This is why I avoided the ice40 parts, which are cheaper and can have high internal memory. The MachXO2 ranges from 388 to 269 MHz. So far I found the penalty to be a factor around 6 for my logic. However there may optimization that I am not doing.

I looked into the MAX10 and Cyclone LP 10. They are nice. I think I favored the MAX10 for some reason despite the Cyclone LP being cheaper and having more internal memory. However I ended up with MACHXO2 since the cost was comparable and it different configuration options.

The parts mentioned here are made by Trenz Electronics, which makes several FPGA ports. Several are available on Digikey or directly from them.

Cellgalvano, 2020-08-09:
I tried using the external ram for some simple tests, but just as you described I also favor the internal BlockRAM, because I only need to wait for one additional clock cycle while reading and my panel setups are pretty small  (32x128 and 64x128). The CYC1000 features 594 kilobits of BlockRAM.

I also tried to do some offloading using a cheap EPM240 CPLD in combination with an asynchronous SRAM, but I could just barely fit my logic and it did not work. Probably a EPM570 will do the trick.

For external RAM I also got some of these IPS6404 QSPI PSRAM ICs laying around which could be interesting because they provide 8MB of RAM with only 6 io pins required.

daveythacher, 2020-08-09:
I just noticed something the ColorLight 5A-75B uses two 16 bit RAM chips with a clock max around 200MHz. Yet supports 8 HUB75 connectors. Meaning that if each connector is driving 6 bit data at 25MHz there is a problem with bandwidth. I want to say the PHYs are 10/100 so the max incoming bandwidth is 200Mbps.

There is two possible explanations. The first is using BCM with the lower three bits dropped. Which would lower cost and power, but be cheating. The other is using a ton of memory, where the input data is expanded into memory. Then played back to HUB75 connectors. That would keep power down too, but increase memory requirements.

The memory trick is better. I had considered this for MCU, but never thought FPGA would need it. BCM could also lower power even further. It would also save memory space too.

marcmerlin, 2020-08-09:
@daveythacher you were asking CPU use.
rpi3 driving 30720 pixels over 3 chains:
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                             
18443 daemon    20   0   25308   4648   3508 R 199.7  0.5   1055:05 Table_Mark_Este  

rip4 driving 98304 pixels over 3 chains:
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND              
 2236 daemon    20   0   27340   6664   3492 R 142.9   0.3 891:14.61 Table_Mark_Este   

as for FPGAs, honeslty if they can't drive 30K pixels or more per chain at let's say 150Hz, then they're not really interesting since an rPi4 can do it at 100Hz or so over 3 chains.

daveythacher, 2020-08-09:
> @daveythacher you were asking CPU use.
> rpi3 driving 30720 pixels over 3 chains:
> PID USER PR NI VIRT RES SHR S %CPU %MEM TIME+ COMMAND
> 18443 daemon 20 0 25308 4648 3508 R 199.7 0.5 1055:05 [...]

I tried to explain in #971. There is a couple tradeoffs that are universal due to HUB75. I question most of this Pi stuff. The HUB75 bandwidth is the limit. Some of what I am hearing is suggesting this is not the case.

The FPGA has some factors to it. The Pi is hard to work with due to the limited external interfaces. However FPGAs are better suited for this. Hardware is almost always better at IO, to my knowledge.

That is some serious CPU usage though. Hzeller mentions this library falls apart when system fully loaded.

How else do those huge LED displays, which these panels are built for, work. They use FPGA slave and master controllers. Everything on the LED panels is ASIC logic.

daveythacher, 2020-08-09:
By the way here is where I got that 1MHz thing from:
>These displays need to be updated constantly to show an image with PWMed LEDs. This is dependent on the length of the chain: for each chain element, about 1'000'000 write operations have to happen every second! (chain_length * 32 pixel long * 16 rows * 11 bit planes * 180 Hz refresh rate).

My reading of this suggests a problem for 30K pixels. 192x192 for example which puts each chain at 64x192 would suggest three bit color at 100Hz. However at 25MHz it would 15 bit color. However I am forgetting this is BCM writes which could make the bandwidth pretty much anything. Although I think it may be 11MHz which would make it 11 or possibly 12 bit color.

daveythacher, 2020-08-10:
> See issue #1038 where I outline a small cheap FPGA. This in theory can do 128x32 @ 24 bit with 47 FPS.

Actually you could do double HUB75 chains with that part. Address and OE would be managed by external processor, but this would allow 128x32 @ 24 bit with 95 FPS. With more pins and memory it could do even more. However for this cost and setup this is not bad. You can bread board this. You could also use multiple chips potentially too. The Pi has two SPI ports. 

SPI kind of ends up being a problem too. However with larger FPGA you could manually bit bang a faster interface. Not recommended but would be more forgiving than this.

marcmerlin, 2020-08-10:
@daveythacher  128x32 at 95fps is just not that interesting, right now I'm doing 128x256 (8 times more) at a 7 bits per pixel
```
            defaults.rows = 64;
            defaults.cols = 128;
            defaults.chain_length = 4;
            defaults.parallel = 3;
            defaults.pwm_lsb_nanoseconds = 100;
            defaults.pwm_bits = 7;
```

To make FPGAs worth it, they ought to be at least 1.5 or 2x faster than a single channel fro the rPi.

daveythacher, 2020-08-11:
> @daveythacher 128x32 at 95fps is just not that interesting, right now I'm doing 128x256 (8 times more) at a 7 bits per pixel with almost the same refresh rate (just a bit faster).
> 
> ```
>             defaults.rows = 64;
>             defaults.cols = 128;
>             defaults.chain_length = 4;
>             defaults.parallel = 3;
>             defaults.pwm_lsb_nanoseconds = 100;
>             defaults.pwm_bits = 7;
> ```
> 
> To make FPGAs worth it, they ought to be at least 1.5 or 2x faster than a single channel fro the rPi.

Alright dude, I am getting a little tried of this. Bump up the PWM bits and actually verify it with a logic analyzer/oscilloscope. Then come talk to me. Because I do not think you understand what I have been telling you. 

Based on your site you are working with 25MHz shift registers with an absolute max of 30MHz @ 4.5-5.5V. Since you are working with FM6126A, The OE by the way may be able to be dropped to 40ns, however that would depend on cable characteristics. Assuming the system/code does not have an issue with it. https://html.alldatasheet.com/html-pdf/1145498/FUMAN/FM6126A/174/1/FM6126A.html 

These are just basic shift registers so there is no PWM or memory mapped driver. Therefore the shift register clock limits the data rate. You have six parallel shift register chains in the HUB75. Half goes to the lower row and the other to the upper. Each color gets its own serial data for each row.

What this means is you are pushing 256 * (2 ^ 7) * (64 / 2) * 100 = 104.8 MHz. That is over three times the listed max. Those datasheets can be conservative, but not that much. More than likely you are not push at 100Hz or you are not pushing 7 bit color even with BCM. BCM has same bandwidth as PWM for LSB. So this calculation still works for part of this. You just do not compute and write for each step.

Please tell me I am wrong if I am missing something here. It would likely be helpful for me. My guess is that it is somehow clipping bits. If I drop the lower two bits of that 7 with BCM you can lose around three percent of the color depth for each color. This is insignificant however allows significant bandwidth reduction with BCM. Remember in BCM the LSBs are critical for bandwidth. That could allow me to be around 26.2MHz.

The issue comes when I try to write 0x3 to the LED, which will always be off. Since you are not actually supporting the full range. I call this cheating. So if you want to compare that then I can do 256x64 @ 24 bit @ 95 FPS. Or how about 512x64 @ 24 bit @ 95 FPS. I mean that is 30K right there in a single chain. You know what I could do more too, but this issue is currently you write 0xF you get 0x0. Overall it ends up being 13.8 million colors instead of 16.7 million.

The entire notion of PWM bits is strange here. It is really BCM, and I am not sure how much of the color depth actually exists. How it does this or determines where to draw the line, I have no idea. However this one thing I came up with to try to explain it.

For this PWM stuff think of windows. A window of time is divided into steps. The lowest resolution of time to express those steps of time is defined by the shift register clock. However we have to send a decent amount of data during this time. Which means we can let certain things slip to by time for a larger data transfer. The data transfer size scales against the column and row.

Larger bits consume the max window, which lowers the refresh rate and limits the data transfer size. However if we let some of this window be incorrect we can tolerate larger data transfers without affecting the bit count or refresh rate. The less significant bits do not need to exist. They will be still present due to magnitude, but their value ignored.

That FPGA I mentioned is cheaper. It is for a personal project which does what I need it too. I went that direction to save money, power and board space. The interesting part of it, has nothing to do here. However the concept can be used by people here. 

The idea is that you can use the FPGA to offload and you can do it for a low cost or without having to get involved in stuff like shields. You also do not need to cripple your software or use a ton of IO pins. Which makes that a heck of a lot more interesting/professional. Overall it would still be unprofessional, but I am not going to bother explaining that again.

I am not going to drop hundreds of dollars for a ton of panels, a power supply(s), etc. I would need a different FPGA board anyhow. Something like that CYC1000/MAX1000 would probably work.

This is the dull way to do embedded Linux for LED display anyhow. I would use a Zynq if I was truly serious about using Linux with LED display. If I was going to use an embedded Linux board I would use BeagleBone Black over Pi anyhow.

marcmerlin, 2020-08-11:
@daveythacher , if I may, too many words, it makes it a bit harder to follow/keep up with everything you're writing :)
To answer your question on Hz at 24bpp or 8 pwm/bcm bits:
```
root@rPi4b:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0 --led-chain=4 --led-parallel=3 --led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=8 --led-panel-type=FM6126A  --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=50 -D4
Size: 512x192. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs                        98.0Hz0Hz 
```
So, yes, I get about 100Hz for 128x256 (led chain 4).

I can't say why it doesn't quite align with the math you posted, but it sounds that basically it's already going as fast or even a bit faster than expected, and therefore I should expect no faster from an FPGA.

daveythacher, 2020-08-11:
> @daveythacher , if I may, too many words, it makes it a bit harder to follow/keep up with everything you're writing :)
> To answer your question on Hz at 24bpp or 8 pwm/bcm bits:
> 
> ```
> root@rPi4b:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular 
> --led-rows=64 --led-cols=128  --led-row-addr-type=0 --led-chain=4 --led-parallel=3 -
> -led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=8 --led-panel-type=FM6126A  
> --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=50 -D4
> Size: 512x192. Hardware gpio mapping: regular
> Press <CTRL-C> to exit and reset LEDs                        98.0Hz
> ```
> 
> So, yes, I get about 100Hz for 128x256 (led chain 4).
> 
> I can't say why it doesn't quite align with the math you posted, but it sounds that basically it's already going as fast or even a bit faster than expected, and therefore I should expect no faster from an FPGA.
> 
> Note that this is also on 3 different channels in parallel, so I'm not even sure if I can get and FPGA that does 3 in parallel, 32K pxels per string, and for much less than $40 while taking framebuffer input over another fast channel so that I can push it from a microcontroller.
> If such a thing exists, with code that is already written, then I'm super interested. If it's just theoretical for now, then it's nice to know it's possible, but until I can buy one and upload the code, it's not super useful to me since I focus on the higher level code that runs on top :)

I was not able to find the Hz print out in the demo applications code, but I did not look very hard. (I would have expected it on new line.) I would appreciate you pointing this out, because I would be interested to see the calculation it uses.

I did not realize you were using three 64x512 chains. I mean you cannot even do 4 bits per color with that and be under 25MHz. The only thing I can think of is you have the slow down set to put it under 25MHz and this is really 3 bits per color for a total 512 colors.

There are many FPGA boards you can get for under 40 dollars that will do this. (See @Cellgalvano's post.) Heck you can by one of the controllers which supports 16 chains. Those are pretty cheap, but the issue with them is you need a transmitter which is more than the receiver. (See previous posts.)

Dude I really cannot tell if you are trolling or not at this point.

marcmerlin, 2020-08-11:
@daveythacher , I'll skip over the questionable accusations
-  --led-show-refresh  shows Hz output
- chain is 4x 128x64 (128x256) on 3 channels, which is 384x256.   Ignore the size output by the lib, it's because it has a virtual mapping and pixels get remapped before display.
- FPGAs that can do it, great, but as I said, until I see one with working code for all my panels, it's just theoretical to me :)

daveythacher, 2020-08-11:
> @daveythacher , I'll skip over the questionable accusations
>
> * --led-show-refresh  shows Hz output
> * chain is 4x 128x64 (128x256) on 3 channels, which is 384x256.   Ignore the size output by the [...]

So the issue I keep making is I keep going back to PWM instead of BCM. PWM makes time easier, but is not required. The fundamental unit of time is derived from it. This library uses OE time for BCM. It has a cross-over point for benefits near the shift register clock divided by rows. This is the fundamental unit of time step for both PWM and BCM.

BCM simply reduces the number of bus transactions. This is a problem in the lower value bits. In the upper value bits it is helpful. PWM keeps this the same. BCM does not, and therefor it enables higher bit values.

512 / 25000000 = 20480ns
(2 ^ 8) * 50ns = 12800ns
(2 ^ 11) * 50ns = 102400ns

Bit plane 0 - 1 * T
Bit plane 1 - 2 * T
Bit plane 2 - 4 * T
Bit plane 3 - 8 * T
Bit plane 4 - 16 * T
Bit plane 5 - 32 * T
Bit plane 6 - 64 * T
Bit plane 7 - 128 * T
Bit plane 8 - 256 * T
Bit plane 9 - 512 * T
Bit plane 10 - 1024 * T

Mine:
Frame_Time = ((2 ^ bit) - 1) * T * ROWS
Refresh Rate = 1 / Frame_Time

According to this:
> max(PWM_LSB_NS * 2^bit, serial_clock_cycle * columns)
T would be either 50ns or 20480ns assuming 25MHz.

I looked into the how the frame rate is measured, it uses a counter. Which if anything would be a high reporting of the time per frame or lower frame rate. If I work backwards from the data you gave it would 409MHz not 25MHz, which I am not sure the Pi can do.

However he does this 11 - PWM_bits stuff. So you are really working from planes 3 thru 10. The really interesting thing is I still think it is really putting out 3 bits of color. In fact this is confirmed by Hzeller's equation and code. I think the serial clock is something like 12.845MHz for your setup.

If I understand this correctly, which I probably don't, then you are bypassing all the waits. It does write while pulser is active, however it is so long that the writes dominate. BCM never takes effect except for maybe in the last bit plane. It does not properly calculate the time step so a hazard/glitch exists.

OE is the time interval for frame rate. OE holds the BCM. If the writes are larger than OE min bit plane time everything breaks down. Thus you effectively start shaving less significant bits. With each bit you lose color depth. When fully compromised you get the number of steps that you have in bits. 

So I think you really have 512. The issue here is software design and max performance of software. I did try to mention this exact thing before.

daveythacher, 2020-08-11:
@hzeller Is there an error if the OE time is less than the serial clock divided by columns?

daveythacher, 2020-08-11:
Here is the some numbers which explains how I got 13MHz.

--led-pwm-bits=8 means we are using the highest 8 bit planes. I am ignoring dithering.

Frame time = 1/98
Row time = Frame time / 32 = 318877.55ns
11 bit plane time = 50 * 1024 = 51200ns
10 bit plane time = 50 * 512 = 25600ns
9 bit plane time = 50 * 256 = 12800ns
...
Row time @ 25MHz = 512/25000000 = 20480ns

At this point it becomes clear that we at least the lower 6 bit planes will be wasted. It is also clear that it cannot waste all 8 bit planes. So I tested wasting 7 bit planes.

(318877.55 - 51200) / 7 = 38239.65ns  - (Is larger than bit plane 10.)
38239.65 / 512 = 74.6868ns
1/74.6868 = 13.389MHz

Now if we apply this to against my equation solving for color depth:
Steps per color: 13389243.8/(512*32*98) = 8.3
Bits per color: log2(8.3) = 3.05
Bits per pixel: 3.05 * 3 = 9.15
Steps per pixel: 2 ^ 9.15 = 568

In summary you are getting the frame rate, but not the color depth.

daveythacher, 2020-08-11:
Do you use inverted OE logic by any chance?

daveythacher, 2020-08-13:
@marcmerlin I would be curious to know if there is a difference with following changes:
- --led-pwm-lsb-nanoseconds=39900
- --led-pwm-bits=3

If I am correct you should see roughly the same image (if not better) and the same FPS. What I did here is properly configure this to match the periods between the two time domains: OE and SCLK. I am only able to do this if I have an estimation of SCK, which I am computing at 12.844MHz.

marcmerlin, 2020-08-13:
@daveythacher  I think the best way to find out, is for you to build it, test it, and report back your findings.
Then you can also build the FPGA solution you've been writing about, compare both, and us know your results.

daveythacher, 2020-08-15:
@marcmerlin Here is what you can expect in terms of performance per chain. It lists rows against columns. It also compares the performance of different panel technology/generations. Note these are generated at max clock and target 95 FPS. However this is probably useful if trying to make huge panels.

This is not useful for this project since SCK is not known or able to be controlled. GCLK is for PWM and MM LED drivers which this project does not support.

`SCK: 25MHz
GCLK: 25MHz-33MHz

FPS: 95+
Multiplex: 1:4
Columns:
Method \ Bit depth	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16
Shift			32768	16384	8192	4096	2048	1024	512	256	128	64	32	16	8	4	2	1
PWM			4096	4096	4096	4096	4096	4096	4096	4096	4096	4096	4096	4096	4096	4096	4096	4096
MM (30 FPS Data)	12288	12288	12288	12288	12288	12288	12288	12288	12288	12288	12288	12288	12288	12288	1288	1288

FPS: 95+
Multiplex: 1:8
Columns:
Method \ Bit depth	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16
Shift			16384	8192	4096	2048	1024	512	256	128	64	32	16	8	4	2	1	0
PWM			2048	2048	2048	2048	2048	2048	2048	2048	2048	2048	2048	2048	2048	2048	2048	0
MM (30 FPS Data)	6144	6144	6144	6144	6144	6144	6144	6144	6144	6144	6144	6144	6144	2048	2048	0

FPS: 95+
Multiplex: 1:16
Columns:
Method \ Bit depth	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16
Shift			8192	4096	2048	1024	256	128	64	32	16	8	4	2	1	0	0	0
PWM			1024	1024	1024	1024	1024	1024	1024	1024	1024	1024	1024	1024	1024	1024	0	0
MM (30 FPS Data)	3072	3072	3072	3072	3072	3072	3072	3072	3072	3072	3072	3072	3072	3072	0	0

FPS: 95+
Multiplex: 1:32
Columns:
Method \ Bit depth	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16
Shift			4096	2048	1024	512	256	128	64	32	16	8	4	2	1	0	0	0
PWM			512	512	512	512	512	512	512	512	512	512	512	512	512	0	0	0
MM (30 FPS Data)	1536	1536	1536	1536	1536	1536	1536	1536	1536	1536	1536	1536	1536	0	0	0`

daveythacher, 2020-08-15:
> @daveythacher I think the best way to find out, is for you to build it, test it, and report back your findings.
> Then you can also build the FPGA solution you've been writing about, compare both, and us know your results.

I have not intention of doing this to be honest. I have looked through the logic. I help clarify things from what I have learned but that is all I intent to do this with this project. 

Looking at the signals with a logic analyzer or scope would be sort of interesting. However this would be me just proving a point. 

I have considered making FPGA open source version. I have toyed around with a decent number of attempts. Most for a personal project that targets a lower resolution than 512x192. Granted many of the topic I use will scale up. Many can be combined into a single solution, Most of my attempts toy around with a few little problems at a time.

My background is in Computer Engineering. I understand a decent amount from certain course. I have had fun putting some of that to use. There are many things that really grind my gears. This project has a few in it. Linux can be useful but I commonly see it wasted or abused. Usually at the expense of other things.

I looked into the OE stuff. It looks like the setting I recommend is not possible. Which is a little unsettling. In fact I am not sure why the OE logic is the way it is. But it has been there since 2015 I want to say.

daveythacher, 2020-08-16:
> I just noticed something the ColorLight 5A-75B uses two 16 bit RAM chips with a clock max around 200MHz. Yet supports 8 HUB75 connectors. Meaning that if each connector is driving 6 bit data at [...]

The other approach is using the on chip memory as buffers. In order for this to work you would need to shift the rows out at a single time. Meaning you pull in all the data for a row into memory. In order to prevent stall you will need to store enough for two. The on chip memory would need to be dual port memory.

Then you need two processes. One is a read process which pulls the next row into the inactive slot of the on chip. Mean while the other process is looping over the row data. For 8 bit per color this means 256 times across the loop. This also becomes the factor for reducing the bandwidth.

On chip memory increases with column size. Bit depth has little influence on this. This is per channel. Memory interface clock is FPS times bit depth per pixel times multiplex ratio divided by memory interface length.

The arbitration factor occurs when using a single memory for dual buffer. You could use multiple memory for each buffer. However this generally prevents copying the data between buffers. Which means the whole frame must be retransmitted. This is not a huge deal since bandwidth generally reserves this amount on the external interface.

However using certain internal architectures you could avoid this and reduce this. Which could allow for stream compression algorithms. The on chip memory acts as cache or buffer against memory interface instability. The question I have is if you can share DDR between a processor and the display. 

DDR works in bursts which may not be stable enough. It has latency for each burst. Here the bursts are likely no large enough. However they are generally designed to work against cache lines. Add too many masters you could have a time issue which bleeds. DDR however is decent for density. Which could reduce cost.

This project uses BCM which in all honesty could be a mistake. It creates to many spikes, but I also think there is a bug in this project related to this. However I see no significant change in the hardware using PWM or BCM.

daveythacher, 2020-08-16:
@Cellgalvano Here is rough outline of what I am talking about.

Using 32 BRAM9 per chain (12.5MHz @ 8 bit memory interface per chain)
	At 1:8 max length is 16384 using 1 bit PWM max
	At 1:16 max length is 8192 using 1 bit PWM max
	At 1:32 max length is 4096 using 1 bit PWM max 

Using 16 BRAM18 per chain (6.3MHz @ 16 bit memory interface per chain)
	At 1:8 max length is 8192 using 2 bit PWM max
	At 1:16 max length is 4096 using 2 bit PWM max
	At 1:32 max length is 2048 using 2 bit PWM max 

Using 8 BRAM36 per chain (3.2MHz @ 32 bit memory interface per chain)
	At 1:8 max length is 4096 using 3 bit PWM max
	At 1:16 max length is 2048 using 3 bit PWM max
	At 1:32 max length is 1024 using 3 bit PWM max 

Using 4 BRAM36 per chain (1.6MHz @ 32 bit memory interface per chain)
	At 1:8 max length is 2048 using 4 bit PWM max
	At 1:16 max length is 1024 using 4 bit PWM max
	At 1:32 max length is 512 using 4 bit PWM max 

Using 2 BRAM36 per chain (0.8MHz @ 32 bit memory interface per chain) <-- Sweet spot
	At 1:8 max length is 1024 using 5 bit PWM max
	At 1:16 max length is 512 using 5 bit PWM max
	At 1:32 max length is 256 using 5 bit PWM max 

Using 2 BRAM36 per chain (0.4MHz @ 32 bit memory interface per chain)
	At 1:8 max length is 512 using 6 bit PWM max
	At 1:16 max length is 256 using 6 bit PWM max
	At 1:32 max length is 128 using 6 bit PWM max
 
Using 1 BRAM72 per chain (0.2MHz @ 64 bit memory interface per chain)
	At 1:8 max length is 256 using 7 bit PWM max
	At 1:16 max length is 128 using 7 bit PWM max
	At 1:32 max length is 64 using 7 bit PWM max 

Using 1 BRAM72 per chain (0.1MHz @ 64 bit memory interface per chain)
	At 1:8 max length is 128 using 8 bit PWM max
	At 1:16 max length is 64 using 8 bit PWM max
	At 1:32 max length is 32 using 8 bit PWM max

Using 1 BRAM72 per chain (0.05MHz @ 64 bit memory interface per chain)
	At 1:8 max length is 64 using 9 bit PWM max
	At 1:16 max length is 32 using 9 bit PWM max
	At 1:32 max length is 16 using 9 bit PWM max

Using 1 BRAM72 per chain (0.025MHz @ 64 bit memory interface per chain)
	At 1:8 max length is 32 using 10 bit PWM max
	At 1:16 max length is 16 using 10 bit PWM max
	At 1:32 max length is 8 using 10 bit PWM max

daveythacher, 2020-08-16:
There are tricks you can use beyond this to reduce on chip memory if willing to increase memory interface clock. There are also tricks you can use work around the different BRAM sizes. 

You can make this simpler and use control logic on processor. It will be more expensive, use more FPGA resource, use a tad more power and be a little slower.  However the bulk of the work is based on the hardware. Which is why I looked into coprocessors. BCM does work in this regard, but it creates problems.

If you can find a nice interface you can use silicon processor between FPGA and CPU. Currently I have only been able to find Zynq for large display setup. (Soft cores also work.) For smaller setups you can make your own for a cost that is not to bad. The design basically follows what I outlined. I worked on a lot of different versions of this.

In summary one can build a FPGA based system that can handle more. It can be done without pure HDL, but pure HDL will be better/professional. This would be reserved for makers, hobbyists, education, and other personal pursuits.

daveythacher, 2020-08-17:
@Cellgalvano It turns out those are on valid ether (L2) frames. There is code which documents the ColorLight-5A-75 some, however it looks incomplete. Are you aware of a complete version? It looks like these are from captures, but it does not look like the capture data is complete enough to find other fields. I know there is features/commands missing.

https://github.com/FalconChristmas/fpp/blob/master/src/channeloutput/ColorLight-5a-75.cpp

Cellgalvano, 2020-08-17:
@daveythacher regarding the ColorLight Boards I'm only following the http://github.com/q3k/chubby75 project. I never thought about using the original firmware, so I can't provide additional information. Sorry.

bluelasers, 2022-03-08:
To answer the question, getting refreshes above 200Hz will be hard at 192x128. Quality will be an issue also. The reason for this is simple, this library uses traditional PWM. Some panels support high refreshes while others do not. FPGAs are used in receiver cards to increase the number of chains possible which increases the serial bandwidth.

Receiver cards also implement another mode called S-PWM. It allows very high refreshes depending on chain length. This is very different from traditional PWM. This more complicated but also generally allows for higher quality also. The Pi could support this however it is only useful for offline computation. This is somewhat involved to properly configure this.

Another option available are SRAM panels, but these are also not supported. The idea that increasing the refresh lowers the brightness, sounds like a configuration problem with this implementation. This can happen, however this may be due to other things. Note there is a limit to how high the refresh can go.

Note the quality is generally at most 11-bits for the LED and increasing the multiple lowers this. You need to make modifications to the code to properly align this. FPGAs are very fast and drive the panels more efficiently than the RPI can, however the cost and complexity is not likely to be worth it. There are specific cases when this is helpful. The most helpful is determinism of the refresh.

In summary you could be asking a bit much of traditional PWM. Lowering the PWM bits will increase the refresh. Adding logic for S-PWM will also increase the refresh without lowering the quality on standard panels. Other than that get a bunch of ESP32s and SRAM panels. Or use a receiver card in net card mode.

There is a lot of application information for these panels that is not mentioned here. However you do not need a FPGA unless you want high quality and/or determinism.

newcokesucks, 2022-10-17:
This code base is designed for low refresh. It can in theory do high refresh with quality and display size reductions. However there will need to be special provisions for ghosting which are not supported by this code base to the best of my knowledge. These provisions are only useful on specific panels. Most panels supported by this project are not designed for high refresh. These panels generally favor color depth over refresh and some favor size over color depth and refresh.

I did some testing an it looks like the size of the display and quality impact stability. The interesting thing is how this code base handles color depth. It assumes 11 bit color depth regardless of multiplex and drops the least significant bits to make more time. This actually creates slight imperfections in accuracy and period times. The effect of this reduces the refresh rate and increases it at the same time. The most significant bits carry the most value but they also carry the most time.

Randomly changing the LSB time will increase the refresh but this will also create errors. There are a number of different things which can affect brightness. I will not take the time to mention that. However it is important to protect the PWM period. This library does not do this and this can be overlooked to a degree. Correcting this in the code would not be very hard if you know what your doing.

There is a discussion for using different LED drivers which by the look of things could be promising for this. However it does not look like any progress has been made. I would say for the time being this project is not rated for high refresh. That being said if you know how to modify certain parameters in the code base you should be able to get a high refresh rate by lowering the color depth. It is believed the stability may be passable for this 192x128, but that is just a hunch.

marcmerlin, 2023-05-10:
bluelasers Joined GitHub on March 8, 2022. No other code written, no other project, no contributions.
newcokesucks Joined GitHub on September 12, 2022 just in time to fork this project and send a CL. No other code written, no other project, no contributions.

If any of you isn't somehow another account for David Tacher, please write your fork with all your ideas and suggestions, and like Linus would say "show us the code".

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828