hzeller/rpi-rgb-led-matrix #919: 64x64 LED Matrix missing lines

iYCloud, 2019-10-29:
Hello!

First of all, thank you for the great library.

Like mentioned in the title, I have a 64x64 panel that has missing lines after every 8x64. I have included a picture down below.

Does anyone have any suggestions about how to get it to work or do I really have a defective panel?

[image]

hzeller, 2019-10-29:
what multiplexing options did you try?

iYCloud, 2019-10-30:
> what multiplexing options did you try?

I have tried all of them `--led-multiplexing=<0..10>` from 0 to 10. And the best result is with 0. Most of them will give the outcome shown on the picture down below.

[image]

hzeller, 2019-10-30:
And it is also not a `--led-panel-type=FM6126A` ?

iYCloud, 2019-10-30:
> And it is also not a `--led-panel-type=FM6126A` ?

From my testing, adding the aforementioned line does not affect the outcome.

iYCloud, 2019-10-31:
I have found a solution to my problem, by connecting the sixth line `GND` to GPIO pin 22 on my Raspberry Pi 3 B+ (following still the wiring diagram [url](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#wiring-diagram)).

My led matrix has a pin layout as follows (picture down below):
`R1 G1`
`B1 GND`
`R2 G2`
`B2 E`
`A B`
`C GND` (I used `GND` as `D`)
`CLK LAT`
`OE GND`

[image]

closed 2019-10-31 by iYCloud

akingscote, 2023-02-17:
Thanks @iYCloud! i had the exact same problem!! cheers