hzeller/rpi-rgb-led-matrix #920: Can this library be used for a custom 96x96 led board?

arahasya, 2019-11-02:
Hi,

Currently we manufacture/assemble LED panels of different sizes.

This is our **new requirement**:

[image]

We are thinking of using 96x96 led boards and multiplex them to achieve the requirement.

Will this library be able to work with it? What changes will I need to make in the code?
Can it be used with a custom led board If I just use the Hub75 with it?

Thanks for this great project.

looking forward for your help!

arahasya, 2019-11-05:
@hzeller can you comment on this please?

hzeller, 2019-12-24:
You have not really given any details about what that custom panel would look like, what controll it is using etc.
Given the parameters that show up in the table (e.g. 1:4 scan), it sounds like it is just like the other panels and you'd be able to control it with this library, maybe with minor modifications.

closed 2019-12-24 by hzeller

arahasya, 2020-01-02:
We will be using 32x16 or 32x32 boards. 
So for a 12 line Display of 384x192, we will require 72 panels of 32x32

It will look like this

 [<][<][<][<][<][<][<][<][<][<][<][<]  }--- Pi connector #1
 [>][>][>][>][>][>][>][>][>][>][>][>]
 [<][<][<][<][<][<][<][<][<][<][<][<]  }--- Pi connector #2
 [>][>][>][>][>][>][>][>][>][>][>][>]
 [<][<][<][<][<][<][<][<][<][<][<][<]  }--- Pi connector #3
 [>][>][>][>][>][>][>][>][>][>][>][>]

The code will look like this for my specifications according to the guide

--led-chain=24 --led-parallel=3 --led-pixel-mapper="U-mapper" ----led-pwm-bits=12 

The boards are currently shipping I will get back to you when I try it out.