hzeller/rpi-rgb-led-matrix #922: Font changes from Panel to Panel

BenJordan23, 2019-11-05:
I have added two new led matrix panels and they are showing another font (distorted font) compared to the first (main) panel.

How can i solve this issue?

[image]
[image]

BenJordan23, 2019-11-23:
Hi @hzeller,
Do you perhaps know how to correct this?
Thank you very much in advance.

hzeller, 2019-11-23:
this looks like some of the a,b,c lines are wired together between the first and second panel. Either the first output panel is broken (test: put the first panel last in chain) or the idc connector between the panels (use a different cable).

BenJordan23, 2020-01-28:
@hzeller , thanks for the reply.

I have 3 panels in total. It seems that 1 panel is a different model. They function correctly when not attached to each other. The daisy chain cables also work fine.

I found out that one model type needs the command "--led-row-addr-type=0" and the other two need "--led-row-addr-type=2" to display the text correctly.

Is there a way to solve this?

FYI here is the back of the two different models:

[image]