hzeller/rpi-rgb-led-matrix #924: 32x16 led - 3chains 2 matrix row problem

Haidar93, 2019-11-10:
Hi,
I am a student and I am building a led wall by using 3 chains with 2 matrices each. I am using a raspberry pi 3.
The problem I am facing when I run the examples is that half of the rows do not work.
I hope you can help me.
Thank you

---

I am tried a lot of different codes:
The baseline was:
--> sudo ./demo -D 1 --led-chain=3 --led-parallel=3 --led-gpio-mapping=regular

hzeller, 2019-11-10:
which half ? Every other ? top/bottom ? stirped ?

Haidar93, 2019-11-10:
[image]

as in the image with 6 matrices but I did not take a picture

hzeller, 2019-11-10:
In your wiring, do you have logic level converters ?
How many lines are there marked A, B, C ?
Do you have a picture of the connector ?
Could it be one with FM6126A chipset ?

Haidar93, 2019-11-10:
I use power supply 5V -10A
I have three lines A,B,C

---
[image]

hzeller, 2019-11-10:
A, B, C sounds expected for a 16 high display.
So I suspect you need level converters from the 3.3V logic to 5V logic (74AHCT245 for instance)
https://github.com/hzeller/rpi-rgb-led-matrix#logic-level-voltage-not-sufficient

Haidar93, 2019-11-21:
I would like to say thank you for Mr. Hzeller for the amazing library.

The problem was in the wire connecting it was connected wrongly

I can not describe my feeling after solved this problem

closed 2019-11-21 by Haidar93