hzeller/rpi-rgb-led-matrix #925: Problem with full brightness of pannell

Danijel777, 2019-11-10:
When RGB matrix is populated with most of "white" LED diodes, picture is pulsating (about 12 Hz). RGB  matrix display is from Coreman (p4 64*32 rgb led matrix panel indoor smd2121 MBI led diplay screen p4 led video wall module p2p3p5p6).
Tho solution of problem is probably in power supply. Voltage droped down for about more then 2 Volts in frequency of about 12 Hz.
In the next days I will take power supplay with posibility of more current. Now I use PC Lenovo power supplay with 5V an max 14 A which is not enough for the peak of current RGB matrix needs.

Here is the link what happend [(https://youtu.be/y4Xf2xjoTL0)]
[image]
[image]

KentWalker, 2019-11-10:
The problem may not be that your power supply cannot deliver enough current but there is not enough current draw from the other supply rails.
PC power supplies expect to see a minimum current draw on both the 5VDC and 12VDC supplies and can become unstable if the load isn't enough.

Try adding some load resistance to the unused supplies.  The LED panel wouldn't draw more than 5A peak so the power supply should be adequate.

hzeller, 2019-11-11:
Sometimes, the spikes in current can also bring the power-supply into some sort of oscillation. I also see that the power cables are pretty long which possibly creates some inductive load (the current pulses needed by the panel are pretty short.

I have seen a 60A power supply oscillate on a panel that only draws 45A average. This was fixed by having local capacitors at the connectors.

I'd try to add some 3300uF low-esr capacitor directly at the power connector on the panel. Here is a good starting point
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power

Danijel777, 2019-11-11:
First, I tried what Mr. Hzeller has said.
The pulsating is smaller but is still there. 
Connection of capacitors is not nice but for evalution purpose is good enough.
Tomorrow I wil try with another PC power supply and consider what KentWalker said.
Thanks to both!
[image]
[image]

Danijel777, 2019-11-12:
Well, with another PC power supplay (connected paralell), everything works fine. 
I have tried with max brightness of whole panel and pulsatings is gone.
With ampermeter I mesaured and white screen, current is about 2,60 A. This is power consumption of 13 W.
I thing that the problem what occur before was in PC power supply what KentWalker said in previous comment.
Possibly, the best solution is to build "classic" rectifier with transformer and 4 diodes in bridge (Graetz) and capacitor (and electronic circuit for stabiliesed voltage). 
PC rectifiers consist of power electronics elements, which sometimes produce unexpected results in case of diferent type of aplication, which was showed in this aplication with RGB LED display.
[image]

Danijel777, 2019-11-14:
One question:
Is the rgb led display I used a good choice or would someone suggest a better manufacturer or product. The purpose is to build a video wall in 3 rows and 4 columns inside or outside the school. So the video wall should be for educational and informational needs. Thank you.

closed 2019-12-18 by Danijel777