hzeller/rpi-rgb-led-matrix #927: 16x16 Matrix with MBI5024 Chipset

ScacchiSL, 2019-11-13:
Hello hzeller

First of all Thank you for your amazing Project and Documantation. I already have a 32x64 Panel working since about a year on my Workbench. :)

Now i have a second project. I had the Opportunity to buy 30pcs 16x16 Led Panels from a Rental Company. Now the Project is to make a Stripe trough the corridor in my Flat.

The new Panels are working on a MBI5024C Chipset and only have Address Line A. I can get some Output with your Software. but i can't get the Mapping in the right order. can you help me out?

Regards
ScacchiSL

ScacchiSL, 2019-11-15:
Found out that MBI5024 is basically the Same than ICN2024

ScacchiSL, 2019-11-15:
[image]
this is the Pinout from the Panel

ScacchiSL, 2019-12-05:
@hzeller any ideas?

hzeller, 2019-12-24:
I don't have the panel, so it is a bit hard for me to figure out the exact internal way it does things. I suspect it might clock in the data with some way of using LAT and Row-A, essentially some new row-address-type that you have to implement. Or they just have a single bit for row and have a long led-multiplexing.

You need to find the datasheet of the controlling chip, then see how it is used in your panel and possibly adapt some code. There is not much I can help without playing with the panels directly.

ScacchiSL, 2019-12-24:
hello @hzeller 

Thx for your reply. i've added the datesheet of the Chip here. If you would like to Play with the panel i can send you one. I'm Based in Switzerland so that wouldn't be a problem to ship. 
[00006155.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/3998870/00006155.pdf)

ScacchiSL, 2020-01-01:
hello @hzeller 

I have imported the Demo Method 12 (Fill All Pixels from @cmagnuson (thx for this amazing tool)

now it was possible to take a video how the LED Panel works.
the following video was taken with this command:
./demo -D12  --led-slowdown-gpio=2 --led-multiplexing=1 --led-rows=16 --led-cols=16

LED1 of the Panel is Upside Left. So the panel works in the wrong order and starts on 16
thirst it starts the LEDs 1 thru 32 the next 32 Steps are the Same LEDs but brighter.

maybe this could help you to point me in the right direction to fix mi Issue. 

https://www.youtube.com/watch?v=DOjt-jQOOzI

ScacchiSL, 2020-02-14:
No hint for me on this issue :( ??

hzeller, 2020-02-14:
It looks like this might have some dual-data-rate situation in which the up and low edge of the clock do something. As I said, I don't have such a panel and won't have time to acquire and test it so you need to test that yourself and find the right bit-pattern and send a pull request.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828