hzeller/rpi-rgb-led-matrix #929: What connectors do you use?

Gansgar, 2019-11-29:
I'm currently in the stage where I need to wire 14 panels together. I would need a 300W power supply to drive them, which I already found. 

But now I'm stuck on which connector I should use to connect the wires (two lanes with 150W each) to the power supply, which I would like to have in a standalone box. I haven't seen any 5V 30A rated connectors yet.

Thanks in advance

hzeller, 2019-12-24:
Anderson Power Pole are usually quite nice connectors, which I use for all my high-amp connections. They come in different sizes, but evern the small ones come with 45A rated contacts.

Gansgar, 2019-12-25:
Ouh nice. That sounds good ^^
I came across them in my research, I just wasn't sure, if they will cut it. 

I'm so happy about your library ^^ I tried to control them with my own library to get a deeper understanding how the panels work and I'm only could appreciate your library more and more. I'm really happy that I don't need to work using the PWM modules and everything, but instead be able to use an abstracted API for that.

Just thank you.

closed 2025-11-27 by marcmerlin