hzeller/rpi-rgb-led-matrix #933: I simply wanted to say thanks for a great library, utilities and hardware designs

The-Oracle, 2019-12-03:
I built a few of your active-3 RPi hats, it was really easy with a framed stencil and stencil printer, and they work splendidly on the 64x64 PH3 indoor panels. I laser cut some 1/4" acrylic to mount 6 panels, 2 panels in parallel with 3 panels chained (192x128), and it makes for an awesome Christmas Pixel Panel display! I wanted to say thanks, because before I found your library, I was considering creating a similar library and app myself from scratch, and that would have been a lot of work! So thank you for sharing this with the community, its a great project! The led-image-viewer is great utility for creating animation streams. Next year I plan to increase the size of the panel to 384x192.

Beersatron, 2019-12-05:
Cool! Do you have picture you can share?

mcpgza, 2019-12-12:
Mine is P2, 256x192, really cool, i love it:

[image]

mcpgza, 2019-12-12:
Another:

[image]

mcpgza, 2019-12-12:
And i'm a big Ori fan:
[image]

The-Oracle, 2019-12-13:
Video was shot through a window, sorry about that.
[image]

hzeller, 2019-12-13:
This is very awesome! Aren't LEDs fun ?

marcmerlin, 2020-01-08:
@mcpgza  what panels to do you use? (I assume 6x 128x64), and how are they connected? (I assume active board with 3x  256x64?). If so, do you get around 150 to 200Hz on them?
Do you use ABCDE panels or AB panels?

marcmerlin, 2020-01-08:
otherwise I think this "issue" can be closed? :)

marcmerlin, 2020-01-08:
That said @mcpgza  I'd love to know if you're getting better refresh than me. I get about 200Hz with video decoding, and the video isn't great, see https://www.youtube.com/watch?v=6yoWvL-_76Y
@hzeller is it expected to have the visible refresh bars at 200Hz on my panels when taking a video in dark conditions (which should select a slower shutter on my camera and therefore make it less likely to see refresh bars)

hzeller, 2020-01-08:
Video decoding is not really recommended on the Pi: it is slow and eats a lot of memory bandwidth, with the result of flickering everything. Pi4 might be slightly better.
Better pre-decode it as stream (yes, that would be uncompressed) or send via network, e.g. using the flaschentaschen-protocol; vlc natively supports that protocol.

The camera adjusts the shutter to the bright parts, and LEDs are very bright :) If you are doing this with a cell phone with limited adjustment-settings, try filming through sunglasses

closed 2025-11-27 by marcmerlin