hzeller/rpi-rgb-led-matrix #934: Issue with outpun on 64x32 P5 8S (scan 1:8) panel

KorneevVitaly, 2019-12-08:
Hello,
I have a problem with output on 64x32 P5 8S RGB panel:

Configuration:
[image]
Output:
[image]

Then I tried demo, and what I've got:
1 Configuration:
./demo -D3  --led-rows=32 --led-cols=16 --led-chain=4 --led-brightness=20 --led-multiplexin g=5 --led-slowdown-gpio=2
[image]

2 Configuration
sudo ./demo -D3
--led-rows=32 --led-cols=16 --led-chain=4 --led-brightness=20 --led-multiplexing= 3 --led-slowdown-gpio=2
[image]

The output always get shifted. Don't know what is wrong with settings.
Please, can you help me?

I use this board: https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/
Panels, that are used:
[image]
[image]

KorneevVitaly, 2019-12-08:
With this configuration is more close, but there ara a lot of errors with it:
 sudo ./demo -D3
--led-rows=32 --led-cols=16 --led-chain=4 --led-brightness=20 --led-multiplexing= 10 --led-slowdown-gpio=2
[image]
[image]

Dtsoft, 2019-12-08:
I am also very interested in solving a similar problem. Thanks.

KorneevVitaly, 2019-12-08:
Tried to write own mapper, but still stuck on wrong output, like it has it's own mapping, that don't match any mapper from this library:
sudo ./demo -D3 -- led-rows=32 --led-cols=64 --led-chain=1 --led-brightness=20 --led-slowdown-gpio=2 --led-multiplexing=9
[image]
Here how mapper now look:
```
 class P10MapperZ : public MultiplexMapperBase {
 public:
   P10MapperZ() : MultiplexMapperBase("P10-128x4-Z", 1) {}
   void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int A_x[64] = {0,1,2,3,8,9,10,11,16,17,18,19,24,25,26,27,32,33,34,35,40,41,4
    2,43,48,49,50,51,56,57,58,59,64,65,66,67,72,73,74,75,80,81,82,83,88,89,90,91,96,
    97,98,99,104,105,106,107,112,113,114,115,120,121,122,123};
   int B_x[64] = {4,5,6,7,12,13,14,15,20,21,22,23,28,29,30,31,36,37,38,39,44,45
    ,46,47,52,53,54,55,60,61,62,63,68,69,70,71,76,77,78,79,84,85,86,87,92,93,94,95,1
    00,101,102,103,108,109,110,111,116,117,118,119,124,125,126,127};

     if (y < 8) {
         *matrix_x = B_x[x];
         *matrix_y = y;
     }
    if ((y>=8) && (y<=15)) {
         *matrix_x = A_x[x];
         *matrix_y = y - 8;
     }
     if ((y>15) && (y<23)) {
         *matrix_y = y + 8;
         *matrix_x = B_x[x];
     }
     if (y>=23) {
         *matrix_y = y;
         *matrix_x = A_x[x];
     }
}
};
```

KorneevVitaly, 2019-12-09:
So, I've edited existing P10MapperZ to look like this:
```
class P10MapperZ : public MultiplexMapperBase {
public:
  P10MapperZ() : MultiplexMapperBase("P10-128x4-Z", 1) {}
  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
   
    if (y < 8) {
        *matrix_x = x + (4+(4*int(x/4)));
        *matrix_y = y;
    }
    if ((y>=8) && (y<=15)) {
        *matrix_x = x + (4+(4*int(x/4))) - 4;
        *matrix_y = y - 8;
    }
    if ((y>15) && (y<23)) {
        *matrix_y = y + 8;
        *matrix_x = x + (4+(4*int(x/4)));
    }
    if (y>=23) {
        *matrix_y = y;
        *matrix_x = x + (4+(4*int(x/4))) - 4;
    }
   
    panel_cols_ = 256;
    panel_rows_ = 16;    
  }
};
```
I used this options:
[image]

And the output now look normal:
[image]

Hope this would be helpfull.

KorneevVitaly, 2019-12-09:
But there is a noise problem. What can be done with it?

KorneevVitaly, 2019-12-18:
Well, the problem with adressing is still present on longer chains:
[image]

KorneevVitaly, 2019-12-18:
So, after some tries and errors, I've come up with this solution for those matrix multiplexor (for now, only for 2 parallel):
```
class WavyCrp : public MultiplexMapperBase {
 public:
  WavyCrp() : MultiplexMapperBase("WawyCrp", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {

    if (y < 8) {
      *matrix_x = x + (4+(4*int(x/4)));
      *matrix_y = y;
    }
    if ((y>=8) && (y<=15)) {
      *matrix_x = x + (4+(4*int(x/4))) - 4;
      *matrix_y = y - 8;
    }
    if ((y>15) && (y<=23)) {
      *matrix_x = x + (4+(4*int(x/4)));
      *matrix_y = y - 8;
    }
    if (y>23) {
      *matrix_x = x + (4+(4*int(x/4))) - 4;
      *matrix_y = y - 8*2;
    }
  }  
};
```

closed 2019-12-18 by KorneevVitaly