hzeller/rpi-rgb-led-matrix #935: 64x64 P3 1:16 scanrate --led-chain is not working in FM6126

gauravoffice96, 2019-12-11:
Configuration:-
sudo ./demo -t 10 -D 0 --led-rows=64 --led-cols=64 --led-panel-type=FM6126A --led-slowdown-gpio=5 --led-multiplexing=1 --led-chain=2 runtext16.ppm
:128x64: Hardware GPIO Mapping

[image]

closed 2019-12-14 by gauravoffice96

hzeller, 2019-12-14:
Did you get it to work ?

gauravoffice96, 2019-12-16:
No it doesn't work,When ever I am passing --led-chain=2 that second panel is not work

[image]

gauravoffice96, 2019-12-16:
i am doing without hat or bonnet.

davemaster, 2019-12-16:
power on second panel?

gauravoffice96, 2019-12-17:
Power is given to second panel even though is not working but
if i increase the --led-cols=128 it is working

[image]

my aim is to make 192x64 panel if i increase the --led-cols=192 this is happening... no multiplexing is supporting
thnx in advance 

[image]