hzeller/rpi-rgb-led-matrix #936: 64x64 p2.5 Adafruit RaspberryPi Problem.

Fabi0220, 2019-12-11:
Good day lovely fellows,
I've got a Problem with my Matrix. 

Technical Details:
64x64 p2.5 Matrix
RaspberryPi 3 
Adafruit RGB Bonnet. 
5V 10amp Power supply

I've already solder jumped pin 4/16 and made a solder for pin 8. 

But now I've got the following problem:
The moment I'm connecting the RaspberryPi with the power supply several LEDs are turning on.

Starting the Demo program turns the Matrix in a full white display. 

I have to tell you, that the Matrix has akready worked, but due to technical problems with the OS I had to reset the RaspberryPi. 

I hope you can help me and for further informations feel free to ask me. 

Pictures will follow this evening. 

Thank you for your effort.

hzeller, 2019-12-16:
Could this be a FM6126A panel ? In that case try to supply the `--led-panel-type=FM6126A` option.

hzeller, 2019-12-24:
Did you get this fixed ?