hzeller/rpi-rgb-led-matrix #937: HUB75 pinout

Danijel777, 2019-12-13:
I recently acquired 18 RGB 16x32 panels for a video project at school. The supplier claims to be the HUB75 standard.
The pin layout is as pictured. Twice the pin R1 is specified. Also no line D. Four lines are marked N (I assume it's GND). Has anyone encountered this kind of pin layout. Or is the pin layout wrong?

[image]

Danijel777, 2019-12-13:
II misspelled. Resolution is 32 x 64 pixells.

hzeller, 2019-12-13:
It just looks like a typo on the silk-screen, so the second `R1` meant `G2`.

It can be only A, B, C without D if this is an outdoor panel (but given that there already is a typo on the silk-screen, who knows :) ).

Danijel777, 2019-12-13:
Thanks a lot. Today I will try to connect.

pet, 13. pro 2019. 05:51 Henner Zeller <notifications@github.com> je
napisao:

Danijel777, 2019-12-13:
Problem solved!
Yes, this is an outdoor panel. Typo is the second R1 (this is G2).

What I did:
First, I didn't connect the D wire.
Secondly, I needed to adjust the parameters to display the content.
This unique matrix (64 x 32) is organized in two  32 x 32 matrix. So, I had to put --led-chain = 2

Other parameters I changed:

- --led-row-addr-type = 3
- --led-multiplexing = 2
- - follows-slow-gpio = 2

Complete statement is:
**sudo ./demo --led-cols = 32 --led-rows = 32 --led-slowdown-gpio = 2 --led-multiplexing = 2 -led-row-addr-type = 3 -led-chain = 2 -led-brightness = 100 -D 1 runtext.ppm**

The only minor problem is the flickering of some pixels. I attribute this to the interference between data wires. 

Thank Mr. Zeller for good tips!

Next step is building small video wall of 3x5 matrix. In few days I expect delivering Adafruit RGB Matrix HAT. 

https://youtu.be/iz5QsZ67Mqw
https://youtu.be/Qy9L6IfF51Q

[image]
[image]

closed 2019-12-18 by Danijel777