hzeller/rpi-rgb-led-matrix #939: Backward text

FelipeSarche, 2019-12-15:
Awesome job, I configured the pins to fit my pcb, it works great but the text is backaward, I see this issue in clock and scrolling text 

./scrolling-text-example --led-rows=16 --led-cols=32 -f ../fonts/helvR12.bdf "FELIPE"

[image]

FelipeSarche, 2019-12-15:
I made the pcb with the driver mbi5026

hzeller, 2019-12-16:
Looks like you have an unusual Matrix that has the pixels reversed.

I've just added a 'Mirror' pixel mapper that should fix your situation, see
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#mirror

Do a ``git pull`, recompile everything, and then you should be able to set the option `--led-pixel-mapper=Mirror:H` to flip your screen horizontally.

hzeller, 2019-12-24:
did you get it to work ?

FelipeSarche, 2019-12-24:
I work great job!!!!!!!!!!!!!!!!!!!

hzeller, 2019-12-24:
Cool!

closed 2019-12-24 by hzeller