hzeller/rpi-rgb-led-matrix #940: Composite video output messes with timing?

KLRico, 2019-12-15:
Hello! First, thank you for this awesome library. This is my first RPi project and your extensive documentation has been very helpful.

I am running a 64x64 matrix on a Pi Zero W. I'm aware of the CPU limitations, but I've been baffled by intermittent/random black lines. Basically, it's like a random number of scan lines are occasionally "missed" and you end up with black bars that occasionally flicker on the display. 

After a couple days of troubleshooting I found out that the issue basically disappears when I use HDMI (I was doing everything headless before).  I enabled **hdmi_force_hotplug=**1 in **/boot/config.txt**, and the problem stayed solved.

From what I understand, the Pi switches to composite video output when HDMI is not detected, so my question is: does the composite video mess with some timing? I'm using an Adafruit matrix hat with the PWM mod.

I don't have any issue with leaving HDMI mode forced, so the problem is solved for me, I just figured I'd bring it up in case something can be done about it from the library side. Maybe it'd help someone else in the future. 

Thanks again for the work you've put into this project.

hzeller, 2019-12-15:
Ah, that is an interesting finding. I also always do headless and have seen random timing issues before, but mostly when there is a lot of network traffic and other expensive computation going on.
If the auto-switch to composite generates issues t is probably a good recommendation do do the HDMI setting as you describe.

I don't know how the Pi does the composite video internally, but it is very conceivable that it can mess with the timing. For one, the internal memory bus of the Pi is easily saturated, so if that is doing something where it copies video data manually, that could possibly happen.

I can also imagine that they might use the same timer generation hardwrae as the library is using which could mess with the outcome.

In any case, would you like to create a pull request and write a little paragraph in the https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting section in the README describing the symptom you see (random black lines) and the solution (force hotplug-config) ?

KLRico, 2019-12-15:
For some reason the hotplug trick isn't working at the moment. If I boot with HDMI and disconnect it, the random line issue is fixed, but it comes back after rebooting headless.

Not sure what is different now, I'm pretty sure I was rebooting without HDMI when I thought I fixed it earlier, but I'll have to keep messing with it. I'll experiment some more and incorporate it into the README when I get a solid fix.