hzeller/rpi-rgb-led-matrix #941: Screen Tear With SwapOnVSync

patrickhirsh, 2019-12-15:
Hi everyone,

I'm utilizing FrameCanvas objects and SwapOnVSync() to update my displays. I'm hoping to better understand when exactly the active buffer is swapped with the supplied buffer, and how multiplexing / parallel chains affects this. 

**Here's my current test setup:**
I'm rendering a cube the bounces back and fourth across the screen at a speed of ~100 pixels per second. I'm updating and "swapping" a non-active buffer every update within my main loop (which is currently not locked to a framerate), which means I'm calling SwapOnVSync _very_ frequently. Obviously this is computationally wasteful, but it's relevant for testing purposes. I expect this to update the RGBMatrix object's non-active buffer every time it's called, then, when we've reached the last scan line and vertical sync is achieved, the RGBMatrix object pulls whichever non-active buffer is currently cached (from the most recent SwapOnVSync() call) and swaps it in. If this were the case, I would expect each scan across the entire panel to be reading from the same buffer.

I'm noticing, however, that the cube is tearing (with the bottom half shifted a pixel over). The tear occurs halfway down the panel (64 X 32, so at row 16) regardless of the cube's vertical position on-screen. 

**My question:**
Am I understanding correctly that SwapOnVSync doesn't swap the active buffer until the entire frame has been "scanned"? This is where my understanding of the hardware gets fuzzy, so apologies if I'm miss-understanding some of these concepts - Is is possible that the buffer is being swapped after the first half of the display is updated due to some kind of multiplexing mode?

How can I ensure that a new buffer isn't swapped in until the _entire_ frame is drawn? Does this change when running panels in parallel? (eg. my full setup is a 4x2 64x32 array of panels - I've also noticed this tearing between the two parallel panels).

Some potentially relevant info:
Using a RPI 4 4GB with [this](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/) breakout board. 4x2 panel array at 64x32 per panel, so --led-parallel=2, --led-chain=4.

As always, thank you very much @hzeller for your fantastic library. I can't remember the last time a project has kept me engaged as long as this RGB Matrix has. It's been a fantastic learning experience, so thank you :)

hzeller, 2019-12-15:
Do you have some code showing how you use it ? If you see tearing, you're updating the wrong buffer.

`SwapOnVSync()` waits until the current buffer is finished (entire frame) then atomically swaps the buffer you pass into it to be the new active buffer. It then returns the previously active buffer, so that is inactive, and you can draw into the inactive buffer.

For an examle how to use SwapOnVSync(), I'd suggest to look at [`scrolling-text-example.cc`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/scrolling-text-example.cc).

patrickhirsh, 2019-12-15:
Unfortunately the code is spread across quite a few files, but I'll try to provide the relevant stuff:

_buffer is my FrameCanvas
`_buffer = _matrix->CreateFrameCanvas();`

The only time I write to a buffer is when I "finalize the frame" at the end of my main panel loop. 
`// panel refresh loop
    while (!interrupt_received)
    {
        // refresh the current mode
        _panelModes[_currentMode]->Update();

        // swap current frame on VSync
        Render::FinalizeFrame();
    }`

This is just writing the frame into the buffer:
`void Render::FinalizeFrame()
{
    if (_matrix == NULL) { ErrorHandler::Log("Render", "Matrix is null!", 
        ErrorNum::FATAL_INVALID_MATRIX); }
    for (int y = 0; y < _matrix->height(); y++)
    {
        for (int x = 0; x < _matrix->width(); x++)
        {
	    _buffer->SetPixel(
		FLIP_HORIZONTAL ? (_matrix->width() - 1) - x : x,
		FLIP_VERTICAL ? (_matrix->height() - 1) - y : y,
		(int)(_currentFrame[x][y]->r * _globalBrightness),
		(int)(_currentFrame[x][y]->g * _globalBrightness),
		(int)(_currentFrame[x][y]->b * _globalBrightness));
        }
    }
    _buffer = _matrix->SwapOnVSync(_buffer);
    
    // clear the swapped buffer for the next frame
    Render::Clear();
}`
I'm swapping the buffer for the active buffer just like in `scrolling-text-sample.cc`.

You mentioned SwapOnVSync() "waits until the current buffer is finished (entire frame)". Is this a blocking call? If not, what is the behavior if I were to call SwapOnVSync(), write to the returned buffer, and call it _again_ before the original call triggers a swap? In other words: what happens if I write to the previously active buffer before the buffers are swapped?

If it _is_ a blocking call, then it seems like this should be working just like scrolling-text-sample.cc does, unless I'm missing something else. 

My project is public [here](https://github.com/patrickhirsh/ProjectDMD), but It's kind of a lot to sift through, so I hope what I've provided here gives you enough context :)

Thanks!

hzeller, 2019-12-15:
`SwapOnVSync()` It is a blocking call. So the way you use it above looks generally good.

hzeller, 2019-12-24:
Could you find what the issue was ?

patrickhirsh, 2020-01-03:
Unfortunately, no. It doesn't appear to be an issue in the example code though (runs fine for me) - despite my implementation being seemingly identical. I'll go ahead and close the issue and update if I figure out what I'm doing wrong :)

closed 2020-01-03 by patrickhirsh