hzeller/rpi-rgb-led-matrix #942: Qiang-Li Q3 64x64 panel, now works.

rttgnck, 2019-12-18:
I have spent some time figuring out why my ABCDE Q3F32v4.1 Qiang-Li 64x64 panel does not work correctly with the Adafruit Matrix Hat. I had not tried it direct wire.

After much time I have come up with an address type for the panel that works based on the ShiftRegisterRowAddressSetter.

`class QiangliQ3SixtyFourRowAddressSetter : public RowAddressSetter {
public:
   QiangliQ3SixtyFourRowAddressSetter(int double_rows, const HardwareMapping &h)
    : double_rows_(double_rows),
      row_mask_(h.e | h.d | h.c | h.b | h.a),
      clock_(h.a),
      data_(h.b),
      bk_(h.c),
      en1_(h.d),
      en2_(h.e),      
      last_row_(-1) {
   }
   virtual gpio_bits_t need_bits() const { return row_mask_; }
   
   virtual void SetRowAddress(GPIO *io, int row) {
      row = row+1;
      gpio_bits_t ra = 0;
      
      if (row == last_row_) return;
      
      // so far I have found that these either turn on or off the row, switch Set and Clear to turn off the rows
      // io->SetBits(bk_);
      // io->ClearBits(bk_);
      
      io->SetBits(clock_);
      
      if (row % 8 == 0) {
         io->SetBits(data_);
      } else {
         io->ClearBits(data_);
      }

      io->ClearBits(clock_);
      if (row >= 1 && row <= 8) {
         io->ClearBits(en1_);
         io->ClearBits(en2_);
      }
      if (row >= 9 && row <= 16) {
         io->SetBits(en1_);
         io->ClearBits(en2_);
      }
      if (row >= 17 && row <= 24) {
         io->ClearBits(en1_);
         io->SetBits(en2_);
      }
      if (row >= 25 && row <= 32) {
         io->SetBits(en1_);
         io->SetBits(en2_);
      }
      last_row_ = row;
   }

private:
  const int double_rows_;
  const gpio_bits_t row_mask_;
  const gpio_bits_t clock_;
  const gpio_bits_t data_;
  const gpio_bits_t bk_;
  const gpio_bits_t en1_;
  const gpio_bits_t en2_;
  int last_row_;
};`

hzeller, 2019-12-24:
Are you working on a pull request for this ?

rttgnck, 2019-12-24:
I had taken a break on that project for some other stuff. But I will get to it this weekend.

darrenmlouw, 2019-12-26:
Could you explain the timings for the latch and enable line, or anything out of the ordinary with the HUB75E interface.

rttgnck, 2019-12-26:
This was found through trial and error mostly. I didn't have to change/add
anything outside of this. It's still a bit flickery and changing the usual
settings for either don't help much, and it does still have some partially
lit other lines. Now it's still an ABCDE panel, so only the
--led-row-addr-type was altered. It's not ABCDE applies directly to the
row. It has Sn5266 drivers where A is clk, B is data, C I found just turns
on or off the row. D is EN1, E is EN2. D and E define which of 1-8, 9-16,
17-24, 25-32. B isn't data data, it tells when  to do the switching for
each 8-sets SP5266.

Like I said though, I didn't analyze the CLK or Latch pin on the HUB75E
port, as it's still partly direct address row setter and acts as such, but
by 8s which is why if you don't toggle B each 8 it won't light up the next
8. It's possible the Hubs CLK or lat pin may help with the flicker though,
or further analysis of the C pin.

darrenmlouw, 2019-12-26:
So should pin A be connected to the same clock as the clk on the hub75e

darrenmlouw, 2019-12-26:
So I found the datasheet for the chip, how would one turn row1/out1 on the Iterate to row2/output2 and so out?   [image]

darrenmlouw, 2019-12-26:
Sorry my spelling was extremely bad, was meant to say “how would one turn row1/out1 on, then Iterate to row2/output2 and so on using B/data?

rttgnck, 2019-12-26:
All I did was replace the class ShiftRegisterRowAddressSetter in the framebuffer.cc file  with the above code and remade the library with `sudo make` from the lib folder. Then started the panel with `sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-row-addr-type=1`. You can alrenatively use the DirectRowAddressSetter and replace with row lookup table. I don't have an algorithm for bitwise operations to create one, but I do have a manual table that can be used.

As far as I can tell, the "clock_" pin used in this address setter class is NOT the same clk pin as the HUB75E port. I do not have the panel connected any longer to a logic analyzer. But I believe the clock_ pin (A) for this address setter merely defines when to change to the next 1 of the 8 rows, data_ (B) tells the panel when to switch to the next 8 of 32, and D and E set which sets of 8 are actually lit up. And for C I have not nailed down its function, but appears to be something like enable/disable the lit state of each line being called in the sets of 8.

Try replacing the ShiftRegisterRowAddressSetter with my QiangliQ3SixtyFourRowAddressSetter code (github didn't format it correctly you need all of the originally pasted code, if you use QiangliQ3SixtyFourRowAddressSetter then you will need to edit the switch in framebuffer.cc InitGpio function with a new one for QiangliQ3SixtyFourRowAddressSetter or reset the class name to ShiftRegisterRowAddressSetter.)

rttgnck, 2019-12-26:
To address you most recent comment, B does not set the row, it tells the panel when to switch to the next 8. I believe A changes the row.

I am also using the Adafruit Matrix Bonnet and this panel may work completely different without it, I have not done a direct wire testing.

darrenmlouw, 2019-12-27:
Thank you very much, you have helped me a lot. I am actually not using a rPi, I am trying to get it working on my FPGA.

daveythacher, 2020-09-13:
This may be implemented with #981

RaspberryPiMatrix, 2021-07-30:
I added the info you request to add to framebuffer.cc   Now how do you Make it to work.  Where do i select to use the QiangliQ3 in.  Normally was rpi-rgb-led-matrix/binding/python/samples/ any of the programs. I am making a YouTube Channel (RaspberryPiMatrix) for raspberry pi Zero/3/4 for Matrix Displays.  I got all other Matrix Displays to work. Trying to get this one to work.  If you have a direct link for the files you created to make the QiangliQ3 to work would be appreciated.  Thank You.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin