hzeller/rpi-rgb-led-matrix #943: We cannot display runtext.ppm file in the library

cerenyasa, 2019-12-19:
```
We have a P10 16x32 RGB SMD Outdoor panel with Hub75 interface and Raspberry pi 3B +. We have made the connections according to the connection scheme on your site, but we can not display the runtext16.ppm image in the examples-api-use file even though we have accessed the director. 

Sudo ./demo -D 1 -m 10 --led-rows = 16 --led-row-addr-type = 2 

to display on the screen when we write runtext16.ppm, the code appears to run in the terminal, but it is not reflected correctly.

![WhatsApp Image 2019-12-18 at 12 29 43](https://user-images.githubusercontent.com/59059790/71188413-f8eb6480-2291-11ea-8ca2-170be51efa3b.jpeg)
![WhatsApp Image 2019-12-18 at 12 29 44](https://user-images.githubusercontent.com/59059790/71188415-f983fb00-2291-11ea-8d1e-6ef0ff4371c0.jpeg)
![WhatsApp Image 2019-12-18 at 12 29 43(1)](https://user-images.githubusercontent.com/59059790/71188416-f983fb00-2291-11ea-88d0-e8bb356aaa30.jpeg)

```

hzeller, 2019-12-24:
Please attache the pictures here in the bug, so that it is easier for everyone to see.

Don't leave spaces between flags and values, just one string. So don't do `--led-rows = 16 ` but `--led-rows=16 `

You might have as well an outdoor panel, did you try the various `--led-multiplexing` options ?

cerenyasa, 2019-12-24:
[image]
[image]
[image]

Multiplexing options 0 to 9 have been tried, but the LEDs in the lower half of the panel did not burn at all in the upper half of the demo code can not get the images we want. In the code we write with Python, it cannot find the RPi.GPIO module, so it does not work either. Can we use another method to check if the panel is working correctly from the terminal screen? The matrix we use is 16x32 P10 SMD Outdoor RGB LED panel A, B, C, D has 4 address line with 1/4 scanning.

sudo ./demo --led-rows = 16 --led-row-addr-type = 2 --led-multiplexing = 1 -D0

When we try the code given above, we can display squares and rectangles, but again we can only get results in the upper half line.

closed 2022-11-27 by cerenyasa97

davemaster, 2022-11-27:
--led-multiplexing = 8 works for me for those 16 x 32 P10 panels, at 1/4 scan
 
 maybe, that panel you have is a different scan/pattern... a picture of your panel back