hzeller/rpi-rgb-led-matrix #944: 2 - 128x64 Boards and Characters are out of alignment at roughly line 32...

jbasemvd, 2019-12-19:
Using 2 128x64 Panels Chained. These are ABC typy using --led-row-addr-type=3.
I'm using a 40 pixel font I downloaded from Google. I get something similar to the following on the display panel. I looked through the Issues here but wasn't able to locate anything that sounded similar. 

Thanks in advance

For the letter I: (Not to scale just showing how the alignment is shifted left)

   ****
   ****

   ****
   ****
   ****
   ****
   ****
   ****
   ****
  ****  <-- Shifted Left at line 32 of 64
  ****
  ****

hzeller, 2019-12-24:
It is not really clear from your description and ASCII drawing what you mean with the shifting. But it can well be that this is just due to shift-register issues.
It might well be that your hardware can't deal with the speed of the GPIO. Try a higher gpio-slowdown value.

jbasemvd, 2019-12-24:
Hello Henner,

 

Thanks for sharing you code it’s been a great learning tool for me.

 

Here is an example of the shifting I was talking about. This doesn’t happen when the text is stationary only when scrolling. The first image is the stationary and the second image is scrolling. I’ve tried all of the GPIO speed variations. I’m using a Raspberry Pi 3.

 

Thanks for responding I appreciate your help.

 

Walt

marcmerlin, 2019-12-30:
@jbasemvd  you should include a picture
Also, are you getting the shifting if you use the demos that came with the library?
If ~/rpi-rgb-led-matrix/examples-api-use/demo -D0 works without shifting, it's not the library's fault :)

closed 2025-11-27 by marcmerlin