hzeller/rpi-rgb-led-matrix #945: using GPIO

GnuPP, 2019-12-22:
HI, first of all, thanks for the great work!
I try to use some of the GPIO's (e.g. the E-Line) for different purpose. For example, to show different content on the LED-screen if the pin level is high or low.

For this I tried using wiringpi and pigpio, but on both I have issues. Is there a way to control the GPIO from inside your libs? Or any other hint?

Thank you very much

d3d9, 2019-12-22:
Check out https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/input-example.cc.
If you want to use it in python, you can try using the code from my fork, see https://github.com/d3d9/rpi-rgb-led-matrix/commit/8f2ee05c5aac42dd16f0a87eb5c021155acbe67f.

hzeller, 2019-12-24:
Yes, the  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/input-example.cc example as suggested by @d3d9 is the way to go @GnuPP

closed 2019-12-24 by hzeller