hzeller/rpi-rgb-led-matrix #948: P4-2121-64*32-16S-HL1

JoSchmaltz, 2019-12-25:
I wired [this](https://a.aliexpress.com/cbEiCZeHH)  P4-2121-64*32-16s-hl1 panel (I believe it has an ICN2028BP chipset) and ran the demo scripts both as a chained 32x32 and as a single 64x32 but didn't get any output except for a few colored stripes.

I did deactivate ic2, spi and sound on the pi 3B+ and tried wiring it according to both instructions, @hzeller 's as well as a different one I found in one of the issue discussions
[image]
...
I also played around with all kinds of parameters...no luck. Anyone in Xmas mood to give me a clue?

hzeller, 2019-12-25:
Always wire it up according to the wiring.md file.
Did you make sure to connect to the input not output of the panel? (connectors are identical, only the silkscreen hints at the right one).
Did you add level shifters?

Could also be that this is one of the panels that require some initialization sequence. There is --led-panel-type which currently supports one of of these.

arrancurran, 2024-02-06:
For anyone ending here with the same issue: 

RPi 4B, OS Bullseye 64bit Lite with P4-2121-64*32-16s-hl1 panel, 'regular' wiring (https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c).

--led-rows=32 --led-cols=64 --led-slowdown-gpio=4 --led-panel-type=FM6127