hzeller/rpi-rgb-led-matrix #949: led-image-viewer not working on pi3 buster w/adafruit RGB matrix bonnet

jvc203, 2019-12-26:
Hello,

As described in title I can't get the led-image-viewer to run an animated gif on a standard 32x32 matrix. It's a Pi3 Buster w/Adafruit RGB Matrix Bonnet.

Everything seems installed and wired correctly because I can get a demo running.
[image]

But when I run the led-image-viewer nothing shows.  From the command line it looks like it's reading the gif correctly?
[image]

Any help is greatly appreciated.  Thanks!

hzeller, 2019-12-26:
The terminal shows what's up: you are using the regular hardware mapping. But since you use a non-standard hardware, in this case the adafruit hat, you need to give `--led-gpio-mapping=adafruit-hat` mapping. Not sure why it worked in your demo, it looks like you compiled it in there ?

jvc203, 2019-12-26:
It works!   Yeah, not sure why the demo was working.  I didn't compile the gpio-mapping line in there..  Anyways, thanks for the help and the quick response!  Great library.

closed 2019-12-29 by hzeller

jabulonboaz, 2020-03-14:
> 
> 
> The terminal shows what's up: you are using the regular hardware mapping. But since you use a non-standard hardware, in this case the adafruit hat, you need to give `--led-gpio-mapping=adafruit-hat` mapping. Not sure why it worked in your demo, it looks like you compiled it in there ?

you just saved my life HZeller i had the same issue just had to use hat-pwm so its stable. this issue has nothing to do with buster just the Adafruit bonet or hat. I use the bonet. thanks again