hzeller/rpi-rgb-led-matrix #950: 64x64 Matrix New only showing bright bars

bebon901, 2019-12-27:
I have just got a 64x64 LED Matrix off pimoroni, for Christmas, with a 2.5mm pitch. I cannot find any settings that work, and I cannot decide which type of chip it is, as both settings (--led-panel-type=FM6126A) with and without just produce the same bars. I have tried every multiplexing option, and decided that 0 produces the most lit up board. I would be extremely grateful for any help, as i am new to these boards, and I would really like to get it working. The situation has baffled me, and I cannot see any other options. I have been through forums following what they suggest as well, and no one seems to have the same issues as me.
Here are some images of the situation. The two lines running across the lower left hand corner of the first picture are just cables, sorry, not dead pixels.
[image]
[image]
[image]
Thank you for any help.

bebon901, 2019-12-27:
`sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=64 --led-no-hardware-pulse --led-panel-type=FM6126A --led-multiplexing=0
`
Was the code I used to run it.

hzeller, 2019-12-27:
How did you wire it up - manual using the wiring.md or did you solder up and adapter board ? Is gnd connected properly? Did you try various row-address modes ? Do you have a sharper picture of the input connector, so that the writing is visible?

bebon901, 2019-12-27:
I have wired it up using your specified wiring for one panel. The GND is connected to the raspberry pi as well. The writing around the connector is:
R1 G1
B1 N
R2 G2
B2 E
A B
C D
CK ST
OE N
 I assume that N is ground, however I only connected one ground to the pi.
I tried pretending it was a 32 32 matrix as well, and still no leds lit up in what should look like a picture.

bebon901, 2020-01-01:
I have tried a 32 by 32 matrix, and it works perfectly fine, however, now when i plug in the 64x64 I get colours and like the image above, but it flashing varying colours, any ideas would be greatly appreciated.

hzeller, 2020-01-01:
maybe it is s different type of panel; do you have a datasheet?

bebon901, 2020-01-11:
No, Pimoroni do not supply datasheets due to them not knowing what type of panels they have sent you.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin