hzeller/rpi-rgb-led-matrix #954: QiangLi P8 40x20 panel

GnuPP, 2019-12-30:
Hi,
I have one of the QiangLi P8 panels @JiDudeS suggested in his pull https://github.com/hzeller/rpi-rgb-led-matrix/pull/879 . I tried so much things, but can't get it to work. I'm using the active-3 board.

This is my command line string:  sudo ./demo -D0 --led-rows=20 --led-cols=40 --led-multiplexing=10 --led-gpio-mapping=regular

I also tried all scan-modes and row-addr-types. No readable result. 
Anybody here who got the panel to run (maybe @JiDudeS)?

Any hints whatelse I could try?
thanks!

GnuPP, 2020-01-02:
solved the problem. it was the missing --led-slowdown-gpio statement.

closed 2020-01-02 by GnuPP

fistel, 2024-01-06:
I have the same problem. Some one can help us, please ?