hzeller/rpi-rgb-led-matrix #956: 64x128 LED Matrix with HUB75 (not HUB75E) interface not working

guyverckw, 2020-01-02:
Hi,
  I just got a few matrix of 64 rows, 128 cols using a HUB75 interface like this one : https://cloud.githubusercontent.com/assets/13116908/14502166/40feb680-01aa-11e6-8041-b4d814c6badb.JPG
  I cannot make it work with this library.  
  I used to have other 4 matrix of same resolutions but with HUB75E interface like this one : 
https://4.bp.blogspot.com/-P3auvkuEEO4/XGrywZ_OXaI/AAAAAAAAEBQ/m3ofo82WCHgfBWxDGLXGt17MJC8F_W_WQCLcBGAs/s1600/Hub%2B75%2Be%2Bconnecotr.jpg
  This one works fine.  

I figure out that in HUB75, the D pin was replaced by a GND pin.  How can I make it work with this library?  Thanks a lot.

hzeller, 2020-01-02:
what row-address-type, panel-type and led-multiplexing did you try?

guyverckw, 2020-01-02:
I got the latest commit and use the following:
sudo ./led-image-viewer --led-show-refresh --led-brightness=50 --led-rows=64 --led-cols=384 --led-chain=1 --led-parallel=2 --led-row-addr-type=3 --led-panel-type=FM6126A --led-slowdown-gpio=3 -C  ~/share/Cinnamoroll_a1.gif

at least this make all 3 panels work, but get lots lot flickering once awhile:

https://youtu.be/p69qbPCMBZA

guyverckw, 2020-01-02:
tried this :
sudo ./led-image-viewer --led-show-refresh --led-brightness=50 --led-rows=64 --led-cols=128 --led-chain=3 --led-parallel=2 --led-row-addr-type=3 --led-panel-type=FM6126A --led-slowdown-gpio=3 --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=50 -C  ~/share/Cinnamoroll_a1.gif

and this : 
sudo ./led-image-viewer --led-show-refresh --led-brightness=50 --led-rows=64 --led-cols=328 --led-chain=1 --led-parallel=2 --led-row-addr-type=3 --led-panel-type=FM6126A --led-slowdown-gpio=3 --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=50 -C  ~/share/Cinnamoroll_a1.gif

same problem

any idea how to fix the flickering problem?
[image]
 this is the connector of the new panels with HUB75, ABC addressing only

hzeller, 2020-01-02:
Flickering: read the troubleshooting section; disable 1wire bus in particular. Make sure to have short data cables and stable power (short, thick wires (AWG 8 / 8mm² or thicker) to power supply; for the shown panels you probably need a 30A supply. Also potentially capacitors; see the "word about power" section in the doc)

guyverckw, 2020-01-02:
I think you're right, very likely cable problem.  Let me try to get some short cable.  For the power, I am already using a 40A power supply, however, will test with 2 separate power supply instead of one later.  Thanks for the tips.

guyverckw, 2020-01-02:
one more question: I am currently using the active board with E-pin set to 8 because of the old HUB75E panels.
[image]

For these new HUB75 panel, since they are using only ABC pin addressing, I don't need to set E-pin on the active board, right?

hzeller, 2020-01-02:
yes, you don't need the E-pin anymore

emabiz, 2020-01-06:
Hi,
I think I have one panel similar to your. It is a P2 64rx128c HUB 75 panel.
[image]

I didn't connect pins D and E, only ABC.
I am using a RaspberryPi2.

I tried it for the first time this morning with these params:
sudo ./demo -D 0 -t 10 --led-cols=128 --led-rows=64 --led-chain=1 --led-parallel=1 --led-row-addr-type=3 --led-multiplexing=0 --led-slowdown-gpio=4 --led-brightness=100

It works quite fine, I noticed little artifacts when the square rotates and I changed --led-slowdown-gpio=4 to have the best result.

Did you check if this example is working in your case?

guyverckw, 2020-01-06:
> Hi,
> I think I have one panel similar to your. It is a P2 64rx128c HUB 75 panel.
> [image]
>
> I didn't connect pins D and E, only ABC.
> I am using a RaspberryPi2.
>
> I tried it for the first time this [...]

You're right, --led-slowdown-gpio=4 actually solve the flickering problem, but the refresh rate dropped to about 40Hz.

I then compensate that with --led-pwm-lsb-nanoseconds=50, the refresh rate can reach about 60Hz now.  

Thanks a lot for the suggestion!

guyverckw, 2020-01-06:
BTW, here is how the [3x3 P2 LED Matrix works like](https://youtu.be/ebOP8b2N9vU)

sub-zerohm, 2024-05-13:
Necro bump but I thought I might add that I'm doing the same thing, you have to buy the higher priced ones to get the ABCDE pins.  Thank you for this thread, because I went with the lower priced ones and was confused at only having ABC

Speedy-VI, 2024-06-03:
Hello. I just received 2 x P2 128x64 LED panels from Ali Express and have discovered several issues with them. First, they only have HUB-75 input connectors on them. The HUB-75 output connectors are missing. I was unhappy about that but then I tried to connect them to an Adafruit Matrix Portal S3 and could not get them to work. Looking closer, the HUB-75 pinout silkscreened on the boards shows only 3 address pins - A, B, C, Address pins D and E say GND, exactly like the picture guyverckw posted above.  The driver chips are ChipOne ICND2083S. 

Based on some Google searches, the ICND2083S chips on these panels require 3 Clock pulses while the Latch is high. I don't think I can control them with a Matrix Portal S3. I found some detailed info in github from pixelmatrix that says panels like this can be controlled with a PI depending on what driver chips they use. The ChipOne ICN2038 is possible (one CLK cycle while LAT is high), but the ICN2038S is at the top of their incompatible list. The chips on my panels are ICND2083S (added "D"). I found spec sheets for the ICN2083S and the ICND2083S but they seem to be identical, at least in how they work.

I may have to give up on these LED panels and either try to return them or set them on my OOPS shelf.  Before I give up, I thought I would post here and ask if there is any possibility of getting these panels to work with an Adafruit Matrix Portal S3. It seems that emabiz and guyverckw were able to control these A, B,, C panels with a Pi, so I guess I could try that, but I would like to use the Matrix Portal S3 if possible. Any insights are appreciated. Thanks.

sub-zerohm, 2024-06-04:
> Hello. I just received 2 x P2 128x64 LED panels from Ali Express and have discovered several issues with them. First, they only have HUB-75 input connectors on them. The HUB-75 output connectors are [...]

I had the same issue, in the above comments it says the option you need addr type 3. Hope this helps

Speedy-VI, 2024-06-04:
> I had the same issue, in the above comments it says the option you need addr type 3. Hope this helps, you would only be able to run one panel though if there's no data output to chain them

Thanks for your response. Based on the posts above I think this type of LED panel can be controlled by a PI but I don't think it is possible with the Adafruit Matrix Portal S3 which is what I have. Even if I got it to work, the missing HUB75 output connector is a deal killer.  The product photo on Ali Express clearly shows 2 HUB-75 connectors. The vendor on Ali Express is willing to let me return the panels, I think with no return shipping charge, so that is what I am going to do. Now that I know that most LED panels that are 64x64 or larger now use only 3 address lines and LED driver chips that shift the addresses, I know what to look out for.

I messaged 20 different vendors on Ali Express and found a few that have P2 128x64 panels with 5 address lines and HUB-75E input and output connectors. I asked them to provide photos to prove it. One particularly helpful vendor sent me a very clear photo of a panel with HUB-75E in and out, and 5 address lines, but the product photo at the link they provided to order it only had a HUB-75E input connector. They assured me that the panels they will send me will have both HUB-75E connectors. I ordered 4 of them so now I wait, again. I told them if I do not receive exactly what we discussed I will return them. This hobby requires a lot of patience!

sub-zerohm, 2024-06-05:
> Thanks for your response. Based on the posts above I think this type of LED panel can be controlled by a PI but I don't think it is possible with the Adafruit Matrix Portal S3 which is what I have. [...]

I did some digging myself, and found that the 128x64 with 5 address lines are just 2 of the 64x64 modules screwed to a wider backing.  So those should have 2 power, and 4 data (2in 2 out) and you just chain it to itself as 2

Speedy-VI, 2024-06-06:
> I did some digging myself, and found that the 128x64 with 5 address lines are just 2 of the 64x64 modules screwed to a wider backing. So those should have 2 power, and 4 data (2 in, 2 out) and you just chain it to itself as 2 panels. It would be nice to get a solid source of ABCDE panels with hub75E, but its not so bad right now. Adafruit and some amazon sellers have them. I built a 4x3 array when i saw the Gameboy project and it runs with maybe 1-2 frames of lag....and thats just a raspberry pi4. The misterFPGA would handle it much better but I'm using what already exists. I forked the Gameboy project to fix issues i had, now i want to add the library to snes9x next, then on to Verilog to write a Mister core for it
> 
> I just want to reiterate that those 3 pin addressing panels are still good and can work with the library, for smaller applications, but if you want to build a bigger wall, you'll need the ABCDE addressing. Its just a matter of biting the bullet on the pricing for now, as far as i can see. I experienced the same flicker issue mentioned above when chaining the ABC panels together with the Pi, but they _did_ work well when used singularly. I drove them with the pi3 and used the matrix HAT that required soldering the whole thing together

I didn’t see any panels at Adafruit over 64x64. I bought a P2.5 64x32 panel from them when I bought the Matrix Portal S3. I paid their premium so I  could start with a panel known to work with the Matrix Portal, and so I could get support from them if I had problems. Turns out that panel has green and blue reversed. When I posted about it, someone pointed out a small note at the bottom of the product description that says these panel have a “slight oddity”  that green and blue are reversed. I did not see the note, so it’s my fault, but it still annoyed me. It’s just a test panel that I am not planning to connect to any other panels so I just flip them in the matrix definition. 

P2 Panels - Of the 20 LED vendors I messaged on Ali Express, 3 had P2 128x64 panels with 5 Address lines. I ended up ordering 4 panels made by Muen from The LED Card Store. Their Ali Express ratings are all around 4.5, so not great, but they were quick to respond and very accommodating. They sent a clear pic of a P2 128x64 panel  with HUB-75 in and out and 5 Address lines clearly visible in the pinout printed by the HUB-75 input connector. They even sent a close-up of the HUB-75 connector with little arrows between the pins and the 5 Address line letters in the pinout. “HUB75” was printed by the input connector (not HUB75E), but the pinout was clear. Their price was a little lower than the 2 other vendors that also had them.

One of the product reviews said the ones they got  were two 64x64 panels chained together. I asked them about this and they confirmed that what they are sending me are single 128x64panels. Also, “P2-1515-128x64-32S-S2” is printed on the panel. They shipped today so the wait begins and fingers crossed. FYI – the other 2 places that had them are EverShine LED and XuYang LED. I don’t know if any of them are solid sources.

I am not familiar with the misterFPGA but Googled it. Wow. That is a whole other level above what I am doing with CircuitPython on the Matrix Portal. I am just making lights blink. What you are doing is truly impressive!

sub-zerohm, 2024-06-08:
> I didn’t see any panels at Adafruit over 64x64. I bought a P2.5 64x32 panel from them when I bought the Matrix Portal S3. I paid their premium so I could start with a panel known to work with the [...]

All I've done personally is fork the gear boy project and fix a couple of values.  The real credit goes to hzeller and dak0r for me personally.

I'm thinking 64x64 is what we're limited to, after More research, but you can find them on AliExpress. Really cheap.  I don't feel so bad after buying 12 panels for made of fruit

Speedy-VI, 2024-06-12:
The 4 P2 128X64 LED panels I ordered from Ali Express arrived today. They got here fast, well packed, and in perfect condition. They have HUB75 in and out connectors and the pinout printed on the boards shows 5 address lines.  I have tried all 4 of them and so far I get nothing, not even a flash when I power up. The panels could be bad but I suspect it is an address line problem.

The Matrix Portal S3 guide says for 64x64 panels you must use address line E, and it may need to be connected to Pin 8 or Pin 16 depending on the panel. They connect it to Pin 8 by default but have solder pads so you can switch it to Pin 16. What I don’t understand is what pins they are referring to. The HUB75 pinouts I have found all show that Pin 8 is address line A, Pin 16 is R1, and Pin 9 is address line E. Maybe Adafruit numbers the HUB75 pins differently?

I attached a diagram of the HUB75 pinout I found online. This exactly matches the pinout printed by the HUB75 input connector on the LED panels. Address line E is Pin 9, so I don’t know what Adafruit is referring to when they say address line E needs to be connected to Pin 8 or Pin 16. I have not given up on making these panels work. Can you help me understand this?

[image]

sub-zerohm, 2024-06-12:
It seems like those pins are numbered backwards.  This is what they should look like, but it's possible they changed it.  These are made by companies who want you to buy their receiving and sending cards so the raspberry is kind of undermining their profits but I don't see why because the real money is in the panels anyway.  Unless you're building a huge concert stage, you don't need to have one of those $1,000 boxes

HUB75.Pinout.png (view on web)

sub-zerohm, 2024-06-12:
Sorry for the multiple emails but I just keep remembering stuff.... I'm not using the Matrix Portal S3.  I'm using the bonnet and I also ordered the joy it one from Germany which has three connectors for LEDs instead of just one and it also has a fan built in.

davemaster, 2024-06-12:
> Thanks for your response. Based on the posts above I think this type of LED panel can be controlled by a PI but I don't think it is possible with the Adafruit Matrix Portal S3 which is what I have. [...]

IF in the picture shows x2 (two) HUB75, You should ask the vendor REFUND or, for a neutral review in comments, ask for panels with chaining capability. What vendor was? For keep that bad experience away...

sub-zerohm, 2024-06-12:
> IF in the picture shows x2 (two) HUB75, You should ask the vendor REFUND or, for a neutral review in comments, ask for panels with chaining capability. What vendor was? For keep that bad experience [...]

They state that some panels may have different configurations so I would guess it's kind of a gamble.  I ordered 12 from adafruit and they're all the same

sub-zerohm, 2024-06-12:
I just wanted to also add that if you're building a Good size business sign then you should go with one of their sending units also.  The raspberry Pi is all experimental at this point

Speedy-VI, 2024-06-13:
> IF in the picture shows x2 (two) HUB75, You should ask the vendor REFUND or, for a neutral review in comments, ask for panels with chaining capability. What vendor was? For keep that bad experience [...]

The vendor was WERALED and I am sending those 2 panels back. They were very nice about it.

Speedy-VI, 2024-06-13:
> I just wanted to also add that if you're building a Good size business sign then you should go with one of their sending units also. The raspberry Pi is all experimental at this point. Unless you really know know you're doing....I'm just getting into this as a hobby to play low resolution games on a CRT-sized array. It's really quite stunning when compared to a standard TV. I run them at like 55% brightness and it's still slightly straining, could use a dark panel over the front maybe.
> 
> You get a slight tear every 16 pixels vertically while running across the screen in a sidescroller, but it's not enough to make it unplayable. It's only one pixel that shifts. I believe old CRTs used to do this as well. If you look really closely it's just harder to see
> 
> The engineers who designed the new Space Invaders with LED panels are really impressive. That thing runs with zero flicker and no tearing, a slight ghosting when the credits roll but that could be just from being run 10 hours a day

I am familiar with LED controllers. I work in the AV industry and am doing a project now with a P1.8 LED display that has over 4 million LEDs driven by a Novastar H15 processing frame. Controlling individual LED panels is just for fun.

I build high-end PCs as a hobby. I got into writing code to control LEDs because I was not happy with the RGB controllers on the market.  I am planning to use three P2 128x64 LED panels in a stack, mounted inside the case for my next build. So far, I am not having any luck with panels larger than 64x32 from Ali Express. I found P2 128x64 panels with 5 address lines and HUB75 in and out connectors from another vendor (LED Card Store). They arrived 2 days ago but so far I can't get them to do anything - not even a flash when I power up. These panels use FM6363C LED driver chips which I am researching now. I found a spec sheet in Chinese and am running it through Google Translate. Things are not looking good for controlling these panels with a Matrix Portal S3. I thought I would be OK if the panel has 5 address lines but I think I may end up sending these back too. I could buy six 64x64 panels from Adafruit but they are $50 a pop. Sigh...

sub-zerohm, 2024-06-22:
> I am familiar with LED controllers. I work in the AV industry and am doing a project now with a P1.8 LED display that has over 4 million LEDs driven by a Novastar H15 processing frame. Controlling [...]

The smaller ABC ones may not have the flicker issues when chained, I don't know yet because I haven't tried them.  I ordered one 64x32 but I blew it up not realizing you need to make sure you don't go over the power rating.  I may try some of these as well but the pitch is slightly bigger, not a big deal it's still high def

Speedy-VI, 2024-06-22:
> The smaller ABC ones may not have the flicker issues when chained, I don't know yet because I haven't tried them. I ordered one 64x32 but I blew it up not realizing you need to make sure you don't [...]

I can't chain the ABC panels I got because they also came with only HUB75 input connectors. I have the return set up on Ali Express and am boxing them up today. I will probably also return the 4 x P2 128x64 panels I bought that do have 5 address lines and HUB75E in and out connectors, but use the FM6363C LED driver chips which require a DCLK, GCLK, and Latch. I have about given up trying to find 128x64 panels that can be controlled by an Adafruit Matrix Portal S3 running CircuitPython. It uses an Espressif ESP32-S3-WROOM-1 MCU. From digging around on github I have learned that these days the higher density panels are using Gen-3 LED driver chips that require specific clock timings and do a lot of complicated signal processing. I will either give up on 128x64 panels and use P4 64x32 panels instead or bite the bullet and buy some P2 64x64 panels from Adafruit that will work with the Matrix Portal S3 but are $50 a pop.

sub-zerohm, 2024-06-23:
> I can't chain the ABC panels I got because they also came with only HUB75 input connectors. I have the return set up on Ali Express and am boxing them up today. I will probably also return the 4 x [...]

The ones I found on AliExpress for like $15 worked.  You could always order a bunch of the 128x64 just for the brackets and attach the 64x64 to them, but it seems like a big waste of money. The magnet solution seems to be the best thing for mounting.  I built extrusion rails and it was a headache

Speedy-VI, 2024-06-26:
> 
> The ones I found on AliExpress for like $15 worked. You could always order a bunch of the 128x64 just for the brackets and attach the 64x64 to them, but it seems like a big waste of money. The magnet solution seems to be the best thing for mounting. I built extrusion rails and it was a headache

Do you know if the ICN2037BP and the FM6124 are functionally the same? I have been chatting with Shenzen Coreman about the Coreman P2 64x64 panels that Adafruit sells for $50 each. In the product photo of the back of the panel, I can see that they use INC2037BP chips, which is the same chip that the P2.5 64x32 panel I bought from Adafruit uses and works fine with the Matrix Portal S3. Coreman told me they have sold Adafruit P2 64x64 panels with ICN2037BP but the more recent batches have FM6124 driver chips and they are functionally the same. They have these panels with 5 address lines and FM6124 driver chips for $12.50 each. Even with shipping from China, they are less than half the price of the Adafruit panels. Coreman also told me they have P2 128x64 panels that use the FM6124 chips. I really want P2 128x64 panels and am hoping that the 128x64 panels that Coreman has with the FM6124 chips will work with the Matrix Portal S3.

sub-zerohm, 2024-06-26:
> Do you know if the ICN2037BP and the FM6124 are functionally the same? I have been chatting with Shenzen Coreman about the Coreman P2 64x64 panels that Adafruit sells for $50 each. In the product [...]

Not to be rude but I believe this forum is supposed to be for issues with functionality, not sourcing.  I am guilty as well but just wanted to say it

Speedy-VI, 2024-06-27:
> Not to be rude but I believe this forum is supposed to be for issues with functionality, not sourcing. I am guilty as well but just wanted to say it

Oh OK. I did not mean to flood this thread with my problems finding panels that work with the Matrix Portal S3 and will stop posting about my search. Thanks for all of your help.

dynamaticx, 2025-03-15:
we want to connect max 10 perller and 10 series.  32X32 led(p6) display with rapberry pi 4b. please guide for this connections.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin