hzeller/rpi-rgb-led-matrix #958: 1/64 Scan on 128x64 matrix

thezenox, 2020-01-05:
I got some hoozoe branded P2.5 128x64 1/64 scan panels using LS9929 shift registers (clone of TLC5926 ?) and LS9739 demultiplexer. (panel and connector pinout see below) Does anyone have some information on how 1/64 scan works with LS9739? (could not find any datasheet) 
Must be some kind of serial communication due to additional U_CLK

[image]
[image]

hzeller, 2020-01-05:
Did you try the various `--led-row-addr-type` ? Some of them actually clock in the address serially.

CzokNorris, 2020-01-08:
Got those modules from Ebay, where the seller claims they work with LINSN or Novastar. Instead of returning them, I will try to get them to work for a few hours first. 
Was able to dig up this electrical specification document:
[室内表贴P2.5规格书320x160-64扫.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/4033273/P2.5.320x160-64.pdf)

hzeller, 2020-01-08:
from the spec, these look like standard ABCDE panels, @CzokNorris not the one discussed here.

CzokNorris, 2020-01-08:
The manufacturer seems to insist, that this is an ABCDE addressed panel. But I cant get it to work on any of my systems. Maybe the difficulty is that its 1/64 scan? No matter what I try and which --led-row-addr-type or which multiplex setting I use, it stays completely dark.

hzeller, 2020-01-08:
First things first: how do you connect the panel? Are you using the adapter in this project or did you wire it up without level-shifters?

CzokNorris, 2020-01-08:
I tried both. I have one wired up to a Raspberry Pi 3b with direct wires as in the wiring documents in this repository. This wiring works with the P5 Panels that I use as a contorl. But nothing lights up on the p2.5 panels. I also have RB-MatrixCTRL boards on a different raspberry pi, connected up to an identicel p2.5 panel. Also nothing. 

https://www.joy-it.net/de/products/RB-MatrixCtrl
For the RB-MatrixCTRL I conencted E to pin 8 of the HUB75-E connector by setting the jumper and pin 4 to GND.

CzokNorris, 2020-01-08:
[image]
[image]
[image]
[image]
Here are some additional images.
[室内表贴P2.5规格书320x160-64扫.zh-CN.en.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/4033646/P2.5.320x160-64.zh-CN.en.pdf)
And a machine Translation of the Datasheet.

This is what the manufacturer said: 
"The buffer of the P2.5 module is 74HC245, decoding ic is 9739, constant current driving ic is 9929."
And thats what he sent about the pinout after I asked him why there is no ABCDE on the siklscreen of the board: 
[image]
[image]

CzokNorris, 2020-01-08:
I am more and more suspecting that the tricky part is running the LS9929 chip. Maybe this chip has some different timing. Also the arrangement of the led driver chips looks odd to me. 
There are eight UUx-Chips on the panel. And eight sets of three led driver chips marked UR UG and UB respectively. 
So it seems like the UUx chips are selecting a set or RGB drivers. But their arrangement looks very odd.

hzeller, 2020-01-08:
maybe we need to initialize the decode chip similar to how FM6126 don't work without. So I guess you have to find a datasheet, make it work with your panel, then send a pull request

CzokNorris, 2020-01-08:
All I got for the shift register is: http://47.99.151.247/product/3-1
There is post about this chip on: https://www.mikrocontroller.net/topic/482296
This chip seems to also be used in flexible panels: https://sqleddisplay.en.alibaba.com/product/62229168372-805839514/P3_flexible_led_display_indoor_soft_flexible_led_module_led_display_screen_led_video.html

thezenox, 2020-01-08:
The two 74HC245 are used as a level shifter (my guess) so no need for the active PCB.
@CzokNorris thanks for the Hangzhou link, there actually is the LS9739 ic as well. So it seems its kind of a dual shift register arrangement, one for line scan, one for columns (for each color).
[image]
I'll try to get a datasheet or protocol description from them and build a test setup with a Teensy 4.0 for testing at least one panel.

daveythacher, 2020-09-12:
My first attempt would this: (Random guess)
U_OE -> Global power. R_OE -> Red color correct, G_OE -> Green color correct, B_OE -> Blue color correct. 

Then shift register address implementation. I would try changing the hardware mappings to put U_CLK on A and put either U on B or C. I would put U_OE on OE. Then I would drive R_OE, G_OE, B_OE low.

alsaleem00, 2021-01-04:
@CzokNorris have you managed to deal with this IC? Does it require initialization as FM6216?

alsaleem00, 2021-01-07:
@thezenox have you managed to deal with this LS9929 IC? Does it require initialization as FM6216?

closed 2025-11-27 by marcmerlin