hzeller/rpi-rgb-led-matrix #961: 128x64 black bar through image

Jacksteel777, 2020-01-14:
`sudo ./led-image-viewer --led-no-hardware-pulse --led-rows=64 --led-cols=128 --led-multiplexing=1 --led-row-addr-type=0 Ponny2.gif`
[image]
[image]
[image]
Configuration works great with P5, but with a P2.5 best image I can get is one with a black bar through the middle. Any help with how to edit is greatly appreciated...I am still very new to this.

hzeller, 2020-01-14:
That is the symptom of a panel with E-line but only controlled with ABCD. Check out the instructions in the readme.

Jacksteel777, 2020-01-18:
I see, thank you, it was fixed by joining the E line to to the 8 on the back of the bonnet.

hzeller, 2020-01-18:
Cool! Have fun with your panel!

closed 2020-01-18 by hzeller