hzeller/rpi-rgb-led-matrix #964: FM6127

gohlhausen, 2020-01-15:
There is a new chip on panels now - FM6127
http://superchip.cn/Private/ProductFiles/63694649249960578960607011.pdf
It is in the same family as FM6126a, but doesn't work with current FM6126a initialization sequences.

The FM6127 adds a third register (on 10) and an OpenState Latch register on 5.

Does anyone have a working initialization sequence for this one?

hzeller, 2020-01-15:
and if you get it working, please send a pull request

gohlhausen, 2020-01-31:
It looks like you can use the FM6126A init sequence if you have the FM6127 - you won't enable bad led detection.
On my board there are also FM5266P row decoder ICs that also need to be initialized - this was causing confusion for me on compatibility.

Here is register information for the FM6127/FM6126A:
Register 1 and Register 2 are for both.  Register 3 is for FM6127 only (it enables bad led detection).

[FM6127.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/4140096/FM6127.txt)

hzeller, 2020-02-03:
So does it work with FM6126 or do you have to change something ? Since I don't have a FM6127 I can't test this. If there is anything to change, please send a pull request.

gohlhausen, 2020-02-03:
At this time I can't tell if my panels are bad or if I'm not sending the correct sequence.
The output I get is this:
X position is correct.
Color is correct.
If i output to y of 1,3,5 or 7, i get all 4 at the same time.  2,4,6,8 work properly.
If i output to y of  10, 12, 14 or 16, i get all 4 at the same time..  9,11,13,15 work properly.
This repeats for 17-32,33-48 and 49-64.

Since all is correct except y, I think the problem isn't with fm, but with the sm5266p decoding or a hardware problem with the boards.

I wanted to include the decoding of the fm6127 control panel that I did in the txt file.  I hadn't seen anything like this on the internet for either fm6126a or fm6127.  it looks like registers 1 and 2 are the same for both.

closed 2020-02-03 by gohlhausen

marcmerlin, 2020-10-22:
This does not seem to work on more than one channel, so I opened a bug for that: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1181

daveythacher, 2020-10-23:
@gohlhausen It looks like you forgot to check configuration settings when generating mask. FM6127 only initializes the first chain, since this pretty much always going to be used. However the others are not certain and therefore probably should not be driven like the FM6126 logic does. So I understand why you did it.

However there are users on here that are not super familiar with certain things. This is a simple fix, but this really should have been reviewed pre-merge. There are users which wish to use only a few chains and drive the other chain's pins for other things. It would be nice if this was not trashed.

There are configuration settings which you should be able to access which will tell you which pins need to be set. All you really do is change what is in the bits_on and maybe bits_off. #1181 is incomplete and misleading. Both the FM6126 and FM6127 are affected.

Support for these were just kind of tacked on. (Again something that should have been caught pre-merge.) The primary reason for these chips to be in existence is not used. Which is the current gain control. However all FM612x logic only drives the init logic. Which is not understood. However since these require init logic sequences these play by different rules than the other panels.

Applying full support is possible but will require some work. Why your initial comment for configuration register bit documentation was dropped I am not sure.