hzeller/rpi-rgb-led-matrix #969: Maximum Refresh Rate

WalidAyyoub, 2020-01-20:
Hello Zeller,

I am developing an application where it is required to have a very high refresh rate, the application requires a camera to take a photo at a certain instant of the LED matrix display, and a high refresh rate is required to avoid conflicts with the camera shutter speed, and frame rate to avoid a case where a photo is taken in between transitions (or while scanning) This is an example of what I required to avoid.
https://www.danielbowen.com/2016/03/03/photographing-led-displays/

I have found LED matrix display with relatively high refresh rates at around 3.8kHz, my question is whether there is a hardware or software limitation to achieve such high rates using this library in combination with a controller such as the Active 3.

marcmerlin, 2020-01-30:
@WalidAyyoub  see https://github.com/hzeller/rpi-rgb-led-matrix/pull/971 and https://github.com/hzeller/rpi-rgb-led-matrix/issues/918
The higher the resolution, the less refresh rate you get (also if you have 1/4 panels, i.e. AB panels) instead of ABCD or ABCDE (/16 or /32), the refresh won't look as bad).
What panels/resolution are you seeing 3.8Khz on? Seems easy/possible with 64x32 AB panels, but harder with 128x64 panels, where I only seem to get 500Hz best case, which is 8 times slower than what you're seeing when the resolution is only 4 times less, so I'm not sure where the 2x factor went.

hzeller, 2020-01-30:
3.8kHz you typically see on 32x16 panels with 1:4 multiplexing.

marcmerlin, 2020-01-30:
Right, with a panel that small (16 times smaller than the one I tested), the math adds up.
@WalidAyyoub basically what this means though is that you're only going to get such a high refresh rate by having a CPU/FPGA drive small panels independently. No chaining, no big panels.
That said if you get /4 or /8 panels (AB or ABC), even if you don't get a super high refresh rate, each line will be displayed a longer amount of time, and will be less likely to look flickery when using a camera.

daveythacher, 2020-08-07:
> 3.8kHz you typically see on 32x16 panels with 1:4 multiplexing.

Likely the row speed not frame. This matters for controllers that use E-PWM/S-PWM. This number also is listed with other numbers, which may help explain it. (I had to guess, and still am.)

daveythacher, 2020-08-07:
> Hello Zeller,
>
> I am developing an application where it is required to have a very high refresh rate, the application requires a camera to take a photo at a certain instant of the LED matrix display, [...]

See my posts in #971