hzeller/rpi-rgb-led-matrix #972: rpi0w is very slow and hangs every few seconds

marcmerlin, 2020-01-30:
Has rPi0w been tested? I kindo of works, but not really.

I used the active-3 which clearly isn't really designed for the pi0 form factor, but it plugs in.
root@rPiZero:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0 --led-chain=1 --led-show-refresh   --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A -D0 

The above works, but the square turns very slowly and every few seconds the screen turns dark for 0.5s or so. The display is otherwise perfect in quality when the pixels are lit.

I then tried to light up my 3 panels, which was the original hope for a wearable where the Pi3 is a bit too big, but I now see that I'm likely going to have to reset my expectations.
root@rPiZero:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0 --led-chain=1 --led-show-refresh --led-parallel=3   --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A -D0
the above works, but the square turns even more slowly (several seconds for a one pixel rotation), and the 3 screens still turn fully off every few seconds for a portion of a second.

marcmerlin, 2020-01-30:
Ok, it seems that the screen going back issues come from "--led-pwm-dither-bits=1". If I remove it, the screen doesn't go dark anymore.
The cube does rotate slowly and uneven speed but I don't get full hangs (74Hz refresh)

Re-adding --led-parallel=3 looks the same, except bigger and slower (refresh is still 74Hz)

marcmerlin, 2020-01-30:
Just in case, I did a make clean, and make on the rpi0 in case the binaries were compatible but not optimized in the right way.  
It seems that -D0 was actually a bit faster, but it still hung every few pixels of rotation's worth, but the hang was for less than a second. Still getting the same 74Hz refresh

hzeller, 2020-01-30:
Pi zero is indeed somewhat slow. The project started out with the first Pi, which had the same cpu as Pi0 now, so yes, it was tested :) But i was glad when the newer pi's came out... Use gpio slowdown to 0. Also don't use pwm tricks that worked in the bigger pi's to not saturate cpu; you don't want to go over 90% cpu; there is only one core. 128x64 is indeed not optimal for this device: it has the worst 1:32 multiplexing. To get a somewhat usable performance, you want to ideally use small 32x16 panels with 1:4 multiplexing to build up your 128x64. Panel. They used to be pretty common; not sure if still available. But also outdoor 1:8 multiplexing 32x32 panels should be much better and get some more usable refresh.

hzeller, 2020-01-30:
the turning off happens due to the realtime reservation in the kennel. I switch that off, when there is more than one core, but it is dangerous for one-core cpu's as it might give 100% to the realtime task, locking up everything.

marcmerlin, 2020-01-30:
@hzeller thanks for the reminder that --led-slowdown-gpio defaults to 1, so I added --led-slowdown-gpio=0, although I didn't see a bit difference.
On the plus side the picture is stable, but there doesn't seem to be enough horsepower to compute the cube rotation without very visible hangs (demo -D0).

Maybe an ESP32 is actually better than an rPi0 at this...
In case you were wondering, I'm not trying to save $10 on the Pi, but building a wearable, and the Pi3 is big, I really was hoping to use a rPi0.
That said, maybe I can start by looking for a smaller active3 board, and one that has connectors on its side, instead of its top (so that the HUB75 plugs don't add to the height)

closed 2025-11-27 by marcmerlin