hzeller/rpi-rgb-led-matrix #973: P10 Hub75 first four lines are off

Flubberflunk, 2020-01-30:
Hi there,

I am using a P10 Hub 75 board. With --led-multiplex=3 option I get a kind of right result.
With every other multiplex the result is a mess.

I tried with demo 3 and as you can see, it looks like the first four rows should be flipped and be on the bottom. 

[image]
[image]
[image]

Any one else got this behavior or got any tips for solutions?

Used command : ./demo -D3 --led-rows=16 --led-cols=32 --led-brightness=10 --led-multiplexing=3

Flubberflunk, 2020-01-31:
Here is the x|y coordinate map, which I created.
looks like it's internally a chained  16*16 panel and every third row is a copied one.

Anyone got an idea?

[image]

Flubberflunk, 2020-01-31:
I figured I had to use the options --led-row-addr-type=1  --led-multiplexing=7, which almost gives me a good result. But I am missing one row.

Any suggestions?

[image]

Used command:

./demo -D 3 --led-rows=16 --led-cols=32 --led-row-addr-type=1 --led-brightness=50 --led-multiplexing=7 --led-pixel-mapper="Rotate:90"

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828