hzeller/rpi-rgb-led-matrix #976: P3 64x32 ICN2037 with Raspberry PI 1 model B: no pixel working

masiorama, 2020-01-31:
Hello everybody,
 in the last days I'm playing with my first matrix led screen (bought last December) and after a few tries I am totally confused, and nothing works.

This is my setup:
- P3 64x32 RGB LED Matrix (ICN2037BP, RUC7258, CHIP ONE ICN74HC245)
- Matrix power supply: [url](https://www.amazon.it/gp/product/B07DQKM9P7/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
- Raspbeery PI 1 Model B with its power supply
  - raspbian onboard, fresh install, with 1-wire and audio driver disabled (I play with it via ssh)

For the wiring I followed [this](https://learn.adafruit.com/16x32-rgb-display-with-raspberry-pi-part-2/wiring-the-displays).

I cloned the library and tried with a few examples, like for example:

`./demo -D 0 --led-cols=64`

but not one led turns on.

I tried with a lot of parametrs but nothing seems to work.
The screen is totally blank, and it shows just a second of blue flashing when I power everyting up, and then nothing more...

I'm sorry if these infos are not enough, but I'm a total n00b, this is my first attempt with this kind of stuff.
Thanks in advance!

penguintutor, 2020-02-03:
I've got a lot of experience with Raspberry Pi and electronics, but I'm having a similar problem.

I have a 32x64 matrix which I'm trying to get working using a Adafruit Matrix Bonnet. At one point I thought I may have been sent a dud matrix, but I can get it working with an Arduino, just not the Raspberry Pi. 

I've tried using the Adafruit installer as well as compiling this from source. I've even tried two different Raspberry Pis. I'm guessing there is a conflict with something else on the Pi, but not sure what. I'll keep looking, but if anyone else has come across this problem and found a solution that would be useful.

hzeller, 2020-02-03:
The ICN2037 might well have some configuration needed that we need to clock in there first.

If you find a detailed datasheet you can experiment with your panel to see if there is some code needed like for the FM6126 ( https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L396 )

If you get it working, we can add another `--led-panel-type` that can deal with this. Please send datasheet findings here for people to comment and possibly send a pull request. 

Related bugs to check discussions there; These are also ICN chips, so they might have a lot in common:
  * #927
  *  #874 
  * #778
  * #466  <- Here is the most discussion, probably fruitful to check this out.

Since I don't have such a panel, I can't test it, so you have to make the necessary experimental steps and prepare the code for a pull request.

masiorama, 2020-02-03:
@penguintutor I'd like to try with an arduino (I think I've got an old Arduino Leonardo) to find out at least if the screen is ok. Can you share how you make it work? Thanks in advance.

@hzeller I'm a total noob, but I will try to follow your suggestions to contribute, thanks!

hzeller, 2020-02-03:
I just read your first message again: you are using the adafruit wiring, please don't do that, they use my library but mix things up a little.
Only use the original wiring documentation found here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

penguintutor, 2020-02-03:
@hzeller Thanks for pointing out that wiring. In my case I am using the Adafruit Bonnet which I believe should work using the option --led-gpio-mapping=adafruit-hat
If I connect directly to the Pi I am concerned that it may damage the Raspberry Pi as I believe I may have damaged a MKR Arduino which uses 3.3V the same as the Pi. I believe that without the power supply to the board then it pulls current from the controller and perhaps overloaded the pins (when connected to an Arduino Mega with no 5V supply the LEDs lit but were dim. I powered on the Arduino before powering on the 5V supply; perhaps the 5V needs to be connected first, but that could be a problem if the 5V is accidentally disconnected before the Pi?

The code for the Arduino is standard code which is the same as the other matrix displays so I didn't think it would need additional code. I will try comparing the code to see if there is anything extra that that Arduino is doing, but that's the reason I thought it may be something else interfering with the GPIO.  I know the snd_bcm2835 module has caused me problems with other RGB lights in the past, but I have already blacklisted that.

@masiorama Unfortunately your Leonardo won't run the code. For a 32x32 you need at least a Uno and for the 32x64 you need an Arduino Mega. You may be able to run the 32x32 code on a Uno connected to the 32x64 matrix (in that case the image is duplicated side-by-side). Also avoid the MKR series (as I stated above I think I may have permanently damaged one by connecting that to the matrix). Wiring details for the Mega are at https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/overview . I think the Pi is more suitable for the matrix rather than the Arduino (due to lack of memory on the Arduino), but it was useful to test using the Arduino to prove that it wasn't a faulty matrix.

masiorama, 2020-02-04:
Indeed it was the wiring. I just tried and the screen is working as intended! Thank you everybody for you suggestions and @hzeller for the library!

penguintutor, 2020-02-08:
I've had chance to try it again and it does work with the standard wiring. The library should support the AdaFruit bonnet with the appropriate parameters, but that's not working. I'll take another look at that when I get chance, but for now I've got it working by wiring directly from the RPi to the ribbon connector.
Thanks.

hzeller, 2020-02-08:
Looks like in their tutorial they use a really really old version of this library with a wiring you can select with `--led-gpio-mapping=classic-pi1`

penguintutor, 2020-02-14:
I can now confirm my problem was hardware related. 

I have returned the AdaFruit Matrix Bonnet to the supplier who have confirmed that it isn't working. So it looks like I may have had a dud. 

I'm sticking with driving it directly from the Raspberry Pi as that takes up less space than with the Bonnet installed.

Thanks for your help and for creating the code which is really useful.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828