hzeller/rpi-rgb-led-matrix #977: Horizontal flickering on panels

amantalion, 2020-01-31:
Hi,

Has anyone encountered horizontal flickering (literally horizontal lines going across the screen) on a chain of panels every 20 seconds or so? If so, then how did you resolve the problem?

I have 6 x P5 64x32 panels with 1/8 scan rate and ICN2037. I use —led-multiplexing=1 and a U-mapper.

Thank you.

hzeller, 2020-01-31:
did you disable 1-wire ?

hzeller, 2020-01-31:
if you have these panels on one chain, you might also max out cpu and the kernel might throttle. If you can, put the panels on parallel chains.

amantalion, 2020-02-01:
Hi @hzeller, thank you so much for the quick reply. I checked and it looked like 1-wire was disabled. I tried to do a clean install (of Raspbian Lite) and now I have noise instead of the desired pattern.

I am running the following command: `sudo examples-api-use/demo -D0 --led-rows=32 --led-cols=64 --led-multiplexing=1` and getting the following noise: 
[image]

Have you encountered this issue before by any chance?

hzeller, 2020-02-01:
This looks like that there is something else competing for the GPIO ports. Check out the troubleshooting section in the README. Also, if the GND is not connected between panel and raspberry pi results in something like this, but given that you are using a level-shifter board, that is unlikely.
Often the panels are slower than the speed of the Pi, so adding some `--led-slowdown-gpio` can help.

hzeller, 2020-02-02:
Did you get it to work ? If so: what was the culprit ?

amantalion, 2020-02-02:
It turns out that the second problem was due to the fact that I accidentally used RPi 4, instead of RPi 3. I fixed it on RPi 4 by using ‘led-slowdown-gpio=2’. However, I encountered horizontal ghosting which I couldn’t fix using troubleshooting guidelines. Therefore, I went back to RPi 3 and it completely fixed the second problem with noise and horizontal ghosting.

I am not sure yet, if the fresh install fixed the horizontal flickering problems. I will test/troubleshoot today and report back (will reopen the issue).

amantalion, 2020-02-07:
Tried to follow the debugging steps and still getting some horizontal flickering. 

However, if I use 2 parallel chains of 3 panels instead of one long chain of 6 panels (with a U-Mapper), I don't see flickering when I run the `examples-api-use/demos` (I tested it with `-D 4`). I still have the same problem, if I try to show a custom image using `utils/led-image-viewer`.

ZuzooVn, 2021-06-03:
@amantalion Did you find the solution?

phwebi, 2025-08-12:
I'm also experiencing this issue on the pi4, with 4 64x64 panels (daisy chained). At a loss on what the cause is.

hzeller, 2025-08-12:
@phwebi you need to explain a bit what you tried; if you are at a loss what the cause is, you probably tried a few options and none of them worked

   * Can  you provide the full command line you're using ?
   * Did you play with `--led-slowdown-gpio` with various values to see if the Pi is too fast ? Did it make a difference ?
   * Do you have a [minimal](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#use-minimal-raspbian-distribution) set-up on your Pi ? Anything that uses CPU/memory bandwidth/GPIO ports can disturb.
   * How do you connect the Pi, are you using a level converter or directly wired ?
   * Is the ground firmly connected between ground of the Pi and the Panel ?

closed 2025-11-27 by marcmerlin